ARMmbed/mbed-os #9454: Test case failed with "CMD timeout: set --retcode true" or "The system has been rebooted due to a fatal error" , when a function is called using a instance , while runnning ice-tea on mbed examples

veenaumachagi, 2019-01-22:
BOARD USED: DISCO_L475VG_IOT01A
Steps to Reproduce:
1.Import "mbed-os-example-ble-Beacon" example.
2.Goto the path "mbed-os/TEST-APPS/device/examples/main.cpp" and open main.cpp file
3.Add "#include ble/BLE.h" and create instance for BLE(BLE &ble = BLE::Instance() in a function , try calling the same function from the main
4.Run the command "mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED" and observe the below logs.

**CODE USED: main.cpp**
/*
 * Copyright (c) 2018 ARM Limited. All rights reserved.
 * SPDX-License-Identifier: Apache-2.0
 * Licensed under the Apache License, Version 2.0 (the License); you may
 * not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an AS IS BASIS, WITHOUT
 * WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include <stdio.h>
#include <stdarg.h>
#include "mbed.h"
#include "mbed-client-cli/ns_cmdline.h"
#include "ble/BLE.h"

#ifndef ICETEA_EXAMPLE_ENABLED
#error [NOT_SUPPORTED] Skipping example application.
#endif
/**
 * Macros for setting console flow control.
 */
#define CONSOLE_FLOWCONTROL_RTS     1
#define CONSOLE_FLOWCONTROL_CTS     2
#define CONSOLE_FLOWCONTROL_RTSCTS  3
#define mbed_console_concat_(x) CONSOLE_FLOWCONTROL_##x
#define mbed_console_concat(x) mbed_console_concat_(x)
#define CONSOLE_FLOWCONTROL mbed_console_concat(MBED_CONF_TARGET_CONSOLE_UART_FLOW_CONTROL)

#define SERIAL_CONSOLE_BAUD_RATE 115200

void cmd_ready_cb(int retcode)
{
    cmd_next(retcode);
}

void wrap_printf(const char *f, va_list a)
{
    vprintf(f, a);
	//BLE &ble = BLE::Instance();
	//ble.getGapState();
	//cmd_printf("Instance created");
}

static int cmd_dummy(int argc, char *argv[]) {
  if( cmd_has_option(argc, argv, "o") ) {
	//BLE &ble = BLE::Instance();
	
	cmd_printf("This is o option\r\n");
  } else {
    //tr_debug("Try to write 'dummy -o' instead");
    return CMDLINE_RETCODE_INVALID_PARAMETERS;
  }
  return CMDLINE_RETCODE_SUCCESS;
}

void run()
[... 2937 lines cut ...]
17:53:01.734 | D1 <-- DutThread: For more i
17:53:01.744 | D1 <-- DutThread: == The system has been rebooted due to a fatal error. ==
17:53:01.765 | D1 <-- DutThread: ARM Ltd
17:53:01.808 | D1 <-- DutThread: ++ MbedOS Error Info ++
17:53:01.818 | D1 <-- DutThread: Error Status: 0x80bad handle to und
17:53:01.828 | D1 <-- DutThread: Location:
17:53:01.848 | D1 <-- DutThread: Error Va
17:53:01.861 | D1 <-- DutThread: Current Thread: main  Id: 0x20002344 Entry: 0x80044B3
17:53:01.881 | D1 <-- DutThread: For more info, visit: https://armmbed.g
17:53:01.881 | TC     MainThread: Test 'test_cmdline' FAIL, reason: Aborted by user
17:53:02.111 User aborted test run
17:53:02.111 Test run aborted.
+----------+---------+-------------+-------------+-----------+----------+
| Testcase | Verdict | Fail Reason | Skip Reason | platforms | duration |
+----------+---------+-------------+-------------+-----------+----------+
+----------+---------+-------------+-------------+-----------+----------+
+---------------+---------+
|    Summary    |         |
+---------------+---------+
| Final Verdict |   PASS  |
|     count     |    0    |
|    passrate   |  0.00 % |
|    Duration   | 0:00:00 |
+---------------+---------+
17:53:02.160 Cleanup done.

jupe, 2019-01-23:
@jonikula this does not look correctly. Aborted test should be visible in summary table and final verdict something else than PASS. This looks another bug in icetea side.

jonikula, 2019-01-24:
Could be, will need to check.

jupe, 2019-01-24:
Yep, thats original issue in device side I guess.

closed 2019-10-12 by ciarmcom