ARMmbed/mbed-os #9459: BLE::Gap::startScan() never timeout

pmancele, 2019-01-22:
### Description
Hello,
I'm working on the new Mbed BLE API, using cordio and the [ST bluenrg cordio driver](https://github.com/ARMmbed/cordio-ble-x-nucleo-idb0xa1.git).
I tried on `mbed-os-5.11` and also on `master`

So far its working but I can't realize a scan for a specific duration. Once the ble stack initialized I want to use the following function of Mbed BLE API(as it is done in the Mbed OS periodic advertiser example):

```c++
    /** Start scanning.
     *
     * @param duration How long to scan for. Special value 0 means scan forever.
     * @param filtering Filtering policy.
     * @param period How long to scan for in single period. If the period is 0 and duration
     *               is nonzero the scan will last for single duration.
     *
     * @note When the duration and period parameters are non-zero scanning will last for
     * the duration within the period. After the scan period has expired a new scan period
     * will begin and scanning. This will repeat until stopScan() is called.
     *
     * @return BLE_ERROR_NONE on success.
     *
     * @see EventHandler::onAdvertisingReport to collect advertising reports.
     * @see EventHandler::onScanTimeout when scanning timeout.
     */
    virtual ble_error_t startScan(
        scan_duration_t duration = scan_duration_t::forever(),
        duplicates_filter_t filtering = duplicates_filter_t::DISABLE,
        scan_period_t period = scan_period_t(0)
    );
```
So to scan for 1 second my code looks like that:

```c++
ble::ScanParameters scan_params;
        scan_params.setOwnAddressType(ble::own_address_type_t::RANDOM);

        ble_error_t error = _gap.setScanParameters(scan_params);

        if (error) {
            print_error(error, "Error caused by Gap::setScanParameters\r\n");
            return;
        }

        error = _gap.startScan(ble::scan_duration_t(100));

        if (error) {
            print_error(error, "Error caused by Gap::startScan\r\n");
            return;
        }

        printf("Scanning started\r\n");
```

Unfortunatly it is scanning forever and the event `onScanTimeout` is never triggered ..

Any idea ?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-01-23:
@ARMmbed/mbed-os-pan

pan-, 2019-01-29:
@paul-szczepanek-arm Could you have a look ?

paul-szczepanek-arm, 2019-01-29:
Sure. Which board are you using so I can repro this?

pmancele, 2019-01-29:
We are using a custom board, based on the STM32L496RG + the ST BlueNRG module. We already used Bluetooth successfully on our board a few month ago with older API.

I'm now trying to use the last BLE examples with CORDIO and `mbed-os-5.11`

I also tried the `BLE_GAP` example which alternate scanning and advertising perdiod, and we can also see the problem. Here is the output of `BLE_GAP` example:

> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Scanning started (interval: 2ms, window: 2ms, timeout: 0ms).
> We found a connectable device
> We have scanned for 5999ms with an interval of 4 timeslots and a window of 4 timeslots
> We have been listening on the radio for at least 5997ms
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Scanning started (interval: 100ms, window: 62ms, timeout: 3000ms).
> We found a connectable device
> We have scanned for 5999ms with an interval of 160 timeslots and a window of 100 timeslots
> We have been listening on the radio for at least 3687ms
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Scanning started (interval: 100ms, window: 25ms, timeout: 0ms).
> We found a connectable device
> We have scanned for 5999ms with an interval of 160 timeslots and a window of 40 timeslots
> We have been listening on the radio for at least 1475ms
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Scanning started (interval: 312ms, window: 6ms, timeout: 0ms).
> We have scanned for 5999ms with an interval of 500 timeslots and a window of 10 timeslots
> We have been listening on the radio for at least 118ms
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Advertising started (type: 0x0, interval: [25 : 50]ms)
> We have advertised for 5999ms with an interval of at least 80 timeslots
> We created at least 0 tx and rx events
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Advertising started (type: 0x2, interval: [62 : 125]ms)
> We have advertised for 5999ms with an interval of at least 200 timeslots
> We created at least 0 tx and rx events
> 
> Starting next GAP demo mode
> DEVICE MAC ADDRESS: dc:1f:bb:00:95:65
> 
> Advertising started (type: 0x3, interval: [62 : 125]ms)
> We have advertised for 5999ms with an interval of at least 200 timeslots
> We created at least 0 tx events
> 

We can see that when there is a timeout configured (3000 ms), it is not triggered and the scan demo stopped after 6000 ms like all the demo.

paul-szczepanek-arm, 2019-01-29:
We had a chat and Vincent pointed out that the new API call only works for BLE 5 chips and needs added backward compatibility due to lack of an event generated by the controller. Thank you for finding this issue, will be fixed in next patch and I'll post the branch here for you if you want to grab the fix early.

pmancele, 2019-01-29:
Ok thank you for the investigation

paul-szczepanek-arm, 2019-01-30:
Thank you for the great bug report, this is the fix if you're in a rush but I'd suggest waiting until we test and merge: https://github.com/ARMmbed/mbed-os/pull/9549

closed 2019-01-31 by cmonr