ARMmbed/mbed-os #9461: time(NULL) returns max value of unsigned int

GregorioMarquez, 2019-01-22:
### Description

I imported the RTC Time example into MBED IDE and after compiling it works, however, if I update Mbed-os to the latest revision, time(NULL) returns the value " 4294967295", which is the maximum value of an unsigned integer.

Target: WIZwiki-W7500P.
tools: MBED online IDE
Mbed-os example revision: c9e63f1

Steps to reproduce:
1 - Import https://os.mbed.com/teams/mbed_example/code/time_HelloWorld/ into compiler
2 - Test code, it works
3 - Update Mbed-os to latest revision
4 - Test, time(NULL) returns 4294967295.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x ] Bug

0xc0170, 2019-08-14:
@GregorioMarquez Sorry for the delay. have you found your answer? The version you refernced was 5.5.1 and in January this year Mbed OS was around ~5.10. 

This might be related: https://github.com/ARMmbed/mbed-os/issues/7543

MarceloSalazar, 2020-11-23:
Closing as we can't reproduce

closed 2020-11-23 by MarceloSalazar