ARMmbed/mbed-os #9464: Huge increase (5K flash, 700B static RAM) between 5.10 and 5.11

janjongboom, 2019-01-22:
### Description

In my minimal (no RTOS) blinky I see a huge increase in flash & memory usage between 5.10 (2fd0c5cfbd83fce62da6308f9d64c0ab64e1f0d6) and 5.11 (a8f0c33eaa2c52babff9655417c36f4b5edd54d7) on K64F with GCC ARM. It seems that UART and printf drivers are always loaded.

```
Mbed OS 5.11, K64F, tiny profile
=
| Module                                   |        .text |     .data |       .bss |
|------------------------------------------|--------------|-----------|------------|
| [fill]                                   |      32(+20) |     3(-1) | 2076(-177) |
| [lib]/c_nano.a                           |    406(+318) | 100(+100) |      4(+4) |
| [lib]/gcc.a                              |      832(+0) |     0(+0) |      0(+0) |
| [lib]/misc                               |      248(+0) |     8(+4) |     28(+0) |
| main.o                                   |       56(+0) |     0(+0) |      4(+0) |
| mbed-os/cmsis                            |  1033(+1033) |     0(+0) |      0(+0) |
| mbed-os/drivers                          |      56(+56) |     0(+0) |      0(+0) |
| mbed-os/hal                              |   1300(+174) |     4(+0) |     66(+0) |
| mbed-os/platform                         |   1241(+969) | 256(+256) |  149(+145) |
| mbed-os/targets                          |  5220(+1302) |    12(+0) |  381(+192) |
| mbed-printf/mbed_printf_implementation.o |    980(+980) |     1(+1) |  188(+188) |
| mbed-printf/mbed_printf_wrapper.o        |        4(+4) |     0(+0) |      0(+0) |

Total Static RAM memory (data + bss): 3280 bytes
Total Flash memory (text + data): 11792 bytes

Mbed OS 5.10, K64F, tiny profile
=
| Module           |    .text |  .data |     .bss |
|------------------|----------|--------|----------|
| [fill]           |   12(+0) |  4(+0) | 2253(+0) |
| [lib]/c_nano.a   |   88(+0) |  0(+0) |    0(+0) |
| [lib]/gcc.a      |  832(+0) |  0(+0) |    0(+0) |
| [lib]/misc       |  248(+0) |  4(+0) |   28(+0) |
| main.o           |   56(+0) |  0(+0) |    4(+0) |
| mbed-os/hal      | 1126(+0) |  4(+0) |   66(+0) |
| mbed-os/platform |  272(+0) |  0(+0) |    4(+0) |
| mbed-os/targets  | 3918(+0) | 12(+0) |  189(+0) |
| Subtotals        | 6552(+0) | 24(+0) | 2544(+0) |

Total Static RAM memory (data + bss): 2568(+0) bytes
Total Flash memory (text + data): 6576(+0) bytes
```

This is the `.mbedignore` file for both releases:

```
mbed-os/rtos/*
mbed-os/components/*
mbed-os/features/*
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

Potentially related: https://jira.arm.com/browse/IOTCORE-878

janjongboom, 2019-01-22:
Bisecting found:

```
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[e4525fbea131356e8308bc5490aba3c99d85a8e1] Mbed Fault Handler was originally placed under rtos folder because of its dependency on some of the RTX data structures for capturing the thread info. But with the new error handling in place thread info collection on hardfaults has been moved to Mbed_error handler. There is no point for fault handler implementation to exist under rtos and can be used for RTOS-less builds as well. So moving under platform folder. Also removing some references to RTX data structs like osRtxInfo from fault handler implementation.
```

cmonr, 2019-01-23:
@SenRamakri Fyi

0xc0170, 2019-01-23:
First, this one should address this : https://github.com/ARMmbed/mbed-os/pull/9296

janjongboom, 2019-01-24:
@0xc0170 Does not help for me. On revision 1b1c1085289e178a88fd56e616d079e1215fd9f3

```
Link: mbed-os-example-blinky-no-rtos
Elf2Bin: mbed-os-example-blinky-no-rtos
| Module                                   |       .text |   .data |      .bss |
|------------------------------------------|-------------|---------|-----------|
| [fill]                                   |      34(+4) |   3(+0) | 2076(-20) |
| [lib]/c_nano.a                           |     406(+0) | 100(+0) |     4(+0) |
| [lib]/gcc.a                              |     832(+0) |   0(+0) |     0(+0) |
| [lib]/misc                               |     248(+0) |   8(+0) |    28(+0) |
| main.o                                   |      56(+0) |   0(+0) |     4(+0) |
| mbed-os/cmsis                            |   1033(+12) |   0(+0) |    0(-84) |
| mbed-os/drivers                          |      62(+6) |   0(+0) |     0(+0) |
| mbed-os/hal                              |    1300(-8) |   4(+0) |    66(+0) |
| mbed-os/platform                         |  1241(+112) | 256(+0) |  149(+24) |
| mbed-os/targets                          |   5220(+26) |  12(+0) |   381(+0) |
| mbed-printf/mbed_printf_implementation.o |     980(+0) |   1(+0) |   188(+0) |
| mbed-printf/mbed_printf_wrapper.o        |       4(+0) |   0(+0) |     0(+0) |
| Subtotals                                | 11416(+152) | 384(+0) | 2896(-80) |
Total Static RAM memory (data + bss): 3280(-80) bytes
Total Flash memory (text + data): 11800(+152) bytes
```

Disabling fault handler does help:

```
| Module                                   |       .text |    .data |       .bss |
|------------------------------------------|-------------|----------|------------|
| [fill]                                   |     10(-24) |    4(+1) | 2253(+177) |
| [lib]/c_nano.a                           |    88(-318) |  0(-100) |      0(-4) |
| [lib]/gcc.a                              |     832(+0) |    0(+0) |      0(+0) |
| [lib]/misc                               |     248(+0) |    4(-4) |     28(+0) |
| main.o                                   |      56(+0) |    0(+0) |      4(+0) |
| mbed-os/cmsis                            |    0(-1033) |    0(+0) |      0(+0) |
| mbed-os/drivers                          |      0(-62) |    0(+0) |      0(+0) |
| mbed-os/hal                              |  1130(-170) |    4(+0) |     66(+0) |
| mbed-os/platform                         |   268(-973) |  0(-256) |    4(-145) |
| mbed-os/targets                          | 3944(-1276) |   12(+0) |  189(-192) |
| mbed-printf/mbed_printf_implementation.o |     0(-980) |    0(-1) |    0(-188) |
| mbed-printf/mbed_printf_wrapper.o        |       0(-4) |    0(+0) |      0(+0) |
| Subtotals                                | 6576(-4840) | 24(-360) | 2544(-352) |
Total Static RAM memory (data + bss): 2568(-712) bytes
Total Flash memory (text + data): 6600(-5200) bytes
```

janjongboom, 2019-01-24:
@SenRamakri OK, thanks for the explanation.

screamerbg, 2019-01-28:
@SenRamakri #9296 doesn't fix the issue.

I just tried to compile the pelion-enablement repository with Mbed OS 5.11.2 for NUCLEO_F207ZG and ARM Compiler 5, and I get the following errors:

```
[1548684258.37][CONN][RXD] >>> Running case #5: 'Pelion Bootstrap & Reg.'...
[1548684258.38][CONN][INF] found KV pair in stream: {{__testcase_start;Pelion Bootstrap & Reg.}}, queued...
[1548684267.72][CONN][RXD]
[1548684267.73][CONN][RXD] [SMCC] Error occurred : MbedCloudClient::ConnectNetworkError
[1548684267.73][CONN][RXD] [SMCC] Error code : 6
[1548684267.74][CONN][RXD] [SMCC] Error details : Client in reconnection mode NetworkError
[1548684273.22][CONN][RXD]
[1548684273.23][CONN][RXD] [SMCC] Error occurred : MbedCloudClient::ConnectMemoryConnectFail
[1548684273.23][CONN][RXD] [SMCC] Error code : 9
[1548684273.24][CONN][RXD] [SMCC] Error details : Client cannot connect anymore Memory allocation failed
```

With Mbed OS 5.10.4, I'm able to successfully run all tests with this board.

To reproduce:
```
mbed import pelion-enablement (requires private access, meaning ARM or Partners staff)
cd pelion-enablement 
cd mbed-os
git checkout master
git pull
cd ..
mbed test -t ARM -m NUCLEO_F207ZG -n simple*tests*dev* -v
```

SenRamakri, 2019-01-29:
@janjongboom - I believe the issue mentioned in this issue has been addressed/answered and so closing this issue. Please let us know if you are seeing other issues in this regards, Thanks for reporting this.

closed 2019-01-29 by SenRamakri

0xc0170, 2019-01-30:
> @janjongboom - I believe the issue mentioned in this issue has been addressed/answered and so closing this issue. Please let us know if you are seeing other issues in this regards, Thanks for reporting this.

There's the report from @screamerbg above 2 days ago, this might be a new issue, @SenRamakri can you review?

SenRamakri, 2019-01-31:
@0xc0170 - That's a different issue as I understand, I think that is more of failures to allocate memory at runtime.