ARMmbed/mbed-os #9467: Unable to update the Serial Baud rate in application and always defaults to baud of 9600 on NRF52840_DK.

yogeshk19, 2019-01-23:
Description

- Target : NRF52840_DK
- Toolchain: GCC
- Tools: mbed-cli
- mbed-os: 5.10.4

Attempting to change the Baud Rate to 115200 doesn't take effect as shown in the code below on NRF52840_DK. I have also attached the bin file built using the online compiler. This has always worked prior to mbed os 5.9 and doesn't seem to work past that version of mbed os. The code below was built against mbed os 5.10.4.

```
#include "mbed.h"
 
int main(void)
{   
    //serial debug
    Serial*  m_pc = NULL;
    
    //set the default log level.
    m_pc = new Serial(USBTX, USBRX);
    m_pc->baud(115200);
     m_pc->set_flow_control(SerialBase::Disabled);
    
     printf("BLE Application Version: %s\n","1.0");
    
    if(m_pc != NULL)
    {
        delete m_pc;
        m_pc = NULL;    
    }
}

```
Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

Thanks,
Yogesh

yogeshk19, 2019-01-24:
Thanks @kjbracey-arm for taking the time to respond and for your feedback, I will implement your suggestion.  However I have a question is the UART turned on by default since stdin and stdout are using those pins? I am wondering if in a production scenario if this needs to be turned off, where we don't want to draw any additional power especially on a NRF52840 module or conditionally turn it in ON/OFF?

Thanks,
Yogesh

kjbracey-arm, 2019-01-24:
Oh, and I believe some of the C libraries may not actually properly open stdin or stdout until the first time you make a read or write call, which may stop `default_console()` being called, and hence the UART being initialised.  So you might be okay, depending on toolchain.

yogeshk19, 2019-01-28:
@kjbracey-arm 
I am not sure what I am doing wrong, however updating the platform.stdio-baud-rate to the desired baud rate, in mbed_app.json, doesn't seem to have any effect. Btw I am compiling the project from the online compiler just so you know. 

Also earlier I used to create a Serial object, because not only was I writing out thru' the serial port using say printf statements, however I used to read from the UART as well. How would I access the data flowing into the UART via the default console, since I am not creating the Serial object anymore.

Thanks,
Yogesh

yogeshk19, 2019-02-06:
@kjbracey-arm, printf statements in MBED-OS by default uses the USBTX pins like you mentioned earlier however the custom board we just got doesn't use the USBTX and USBRX pins, however the board has JLink headers that we program the NRF52840. So I was wondering if there is a way to re-direct the printf o/p to use the JLink headers so this can be potentially viewed on JLinkRTTViewer or a JLinkRTTClient application or any other means?

kjbracey-arm, 2019-02-15:
> How would I access the data flowing into the UART via the default console, since I am not creating the Serial object anymore.

`stdin`, so just `getc` or `scanf`.

> So I was wondering if there is a way to re-direct the printf o/p to use the JLink headers so this can be potentially viewed on JLinkRTTViewer or a JLinkRTTClient application or any other means?

Yes, you can override the default console. See `mbed::mbed_override_console` in mbed_retarget.h - more information on that [here](https://github.com/kjbracey-arm/mbed-os-5-docs/blob/5569e244d44cfe8fd7acf38d92085bd179888fca/docs/api/platform/FileHandle.md). I believe the SerialWireOutput is what you need, so just return an instance of that from `mbed_override_console`, similar to the example there.

yogeshk19, 2019-02-24:
Thanks @kjbracey-arm . The changes you suggested works very well. I appreciate you taking the time to help me with this issue. I am closing this issue.

closed 2019-02-24 by yogeshk19