ARMmbed/mbed-os #9490: emac: files do not contain license description

0xc0170, 2019-01-24:
### Description

Files in features/netsocket/emac-drivers/, for instance features/netsocket/emac-drivers/TARGET_NXP_EMAC/TARGET_IMX/imx_emac.h do not contain SPDX identifier neither license description

@ARMmbed/mbed-os-ipcore Please review

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-01-24:
Also features/netsocket/emac-drivers/TARGET_NXP_EMAC/TARGET_LPCTarget/lpc17_emac.cpp - does not contain any license information

SeppoTakalo, 2019-02-25:
@TomoYamanaka features/netsocket/emac-drivers/TARGET_RZ_A1_EMAC/rza1_emac.cpp does not contain Copyright header. Please add.

This is contribution from Renesas, so I cannot claim the copyright to it.

SeppoTakalo, 2019-02-25:
@svastm Please add Copyright header to `features/netsocket/emac-drivers/TARGET_STM_EMAC/TARGET_STM32F4/TARGET_MODULE_UBLOX_ODIN_W2/stm32f4_eth_init.cpp` as the file is originally submitted by you. Thanks.

SeppoTakalo, 2019-02-25:
@ARMmbed/team-ublox Please add copyright header to `TARGET_STM_EMAC/TARGET_STM32F4/TARGET_MODULE_UBLOX_ODIN_W2/wifi_emac/wifi_emac.cpp`

SeppoTakalo, 2019-02-25:
I found only one driver that I can fix, and submitted a PR in https://github.com/ARMmbed/mbed-os/pull/9834

Rest are contributions from mentioned parties, and we should wait for them to submit the Copyright header.

0xc0170, 2019-04-12:
> @svastm Please add Copyright header to features/netsocket/emac-drivers/TARGET_STM_EMAC/TARGET_STM32F4/TARGET_MODULE_UBLOX_ODIN_W2/stm32f4_eth_init.cpp as the file is originally submitted by you. Thanks.

@ARMmbed/team-st-mcd 

> @ARMmbed/team-ublox Please add copyright header to TARGET_STM_EMAC/TARGET_STM32F4/TARGET_MODULE_UBLOX_ODIN_W2/wifi_emac/wifi_emac.cpp

@ARMmbed/team-ublox

jeromecoutant, 2019-04-12:
Hi
@svastm is no more part of ST :-(
but as file is UBLOX one, I let @ARMmbed/team-ublox provide both patches in the same PR
Regards,

0xc0170, 2019-08-14:
This issue is still valid (I checked stm32f4_eth_init.cpp).

@ARMmbed/team-ublox Can you fix license headers in your files please? This issue could be closed once fixed.

cc @maclobdell

aqib-ublox, 2019-08-20:
@0xc0170 sure in next PR

aqib-ublox, 2019-08-20:
https://github.com/ARMmbed/mbed-os/pull/11266 
added

0xc0170, 2019-08-26:
This can be closed, all files were updated.

closed 2019-08-26 by 0xc0170