ARMmbed/mbed-os #9511: Disabling crash-capture feature still allocated memory in RAM

deepikabhavnani, 2019-01-25:
### Description

Crash capture feature can be disabled with config option `"platform.crash-capture-enabled" : 0`, but that does not impact the linker scripts. Memory in RAM is still allocated and all regions are added.

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct#L84

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

CC @SenRamakri

0xc0170, 2019-08-13:
Should the linker scripts be protected by default, aren't they? I've seen today the issue about linker script not getting all config macros, so there might be also issue there. Ideally, it would work (capture disabled, no sections allocation).

cc @ARMmbed/mbed-os-core

deepikabhavnani, 2019-08-13:
@0xc0170 - Apologies, my wordings were not clear. In this case config is not used at all.

Ram memory (0x100) for crash capture is always allocated in linker scripts even when `crash-capture-enabled` is set to disable. 
Linker script should be modified to use this config parameter  and then create a special region. In case of scatter files it could be done by setting `#define m_crash_report_ram_size        (0x0)`