ARMmbed/mbed-os #9512: Cordio BLE: ConnectionCompleteEvent seems to be missing one of the Connection Interval properties in the class implementation?

yogeshk19, 2019-01-28:
Description

Target : NRF52840_DK
Toolchain: GCC
Tools: mbed-cli
mbed-os: 5.11.2

I am using the BLE Cordio API and my custom application I have written is a BLE Peripheral. I subscribe to the onConnectionComplete as shown below as example.  The input parameter is an object of type ConnectionCompleteEvent. Some of the properties in this object I am looking for is the  MinConnectionInterval and MaxConnectionInterval. However this object only contains one property "connectionInterval" and can be retrieved via getConnectionInterval method. So I am not sure if this is a bug or am I missing something here, since I don't see two intervals and I am not sure which Connection Interval property the object refers to (Min or Max)? 

Also if you look in the code below when I want to send an updated connection parameters to the connected peer, I have two intervals I can set in the API.

```
void BleTestApp::onConnectionComplete(const ble::ConnectionCompleteEvent &event)
{
    if (event.getStatus() == BLE_ERROR_NONE) {
        m_bIsConnected = true;
        m_nConnectionHandle = event.getConnectionHandle();
        
        //update the connection parameters to set better power management.
        m_ble.gap().updateConnectionParameters(m_nConnectionHandle,
                                    ble::conn_interval_t(MinConnectionInterval),
                                    ble::conn_interval_t(MaxConnectionInterval),
                                    ble::slave_latency_t(SlaveLatency),
                                    ble::supervision_timeout_t(ConnectionSupervisionTimeout));
        
    }
} 
```

Issue request type
[ ] Question
[ ] Enhancement
[X] Bug

Thanks,
Yogesh

pan-, 2019-01-29:
Hi @yogeshk19 ,

That is not a bug nor a defect in the new API; the connection event doesn't contain a [min, max] interval in the Bluetooth specification so it doesn't contain one in our API. 
Interval is useful when the application request connection establishment or parameter update. It allows the Bluetooth Stack/Controller to select the appropriate value from the range proposed by the application as other radio traffic may happen concurrently.  But in the end, the controller will select a unique value as the connection interval.

yogeshk19, 2019-01-29:
Hi @pan- ,

Thanks for your response and I was wondering is this new in BLE 5 spec? Prior to the mbed os 5.9 or may be even later the connection event returned these two parameters. This was one of the ways I used to confirm if the update in connection parameter request succeeded or not? In the new API structure since 5.11, is there a way to verify if the updated connection parameters request succeeded? 

Thanks,
Yogesh

yogeshk19, 2019-02-05:
@pan- In my case its a peripheral requesting the update to the connection parameters. Typically a custom IPhone app/android app would connect to the peripheral. A while back you had shared a link to Apples BLE connection parameters equation. I have used that successfully in the past to lower the power draw to a few uA's in connected state. So I was hoping to achieve the same with the new API's as well.

So I can subscribe to the ConnectionParametersUpdateCompleteEvent and figure out what the actual parameters were negotiated correct? 

As always thanks a lot for your help. Just curious does Android phones have similar equations to that of Apple that you are aware of?

Thanks,
Yogesh

pan-, 2019-02-05:
Yes, you will receive new parameters - when the update is complete - in your implementation of `onConnectionParametersUpdateComplete`. 

> Just curious does Android phones have similar equations to that of Apple that you are aware of?

I don't think google define any specific value. However Android defines a [function](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)) that allows you to set the connection _priority_.

yogeshk19, 2019-02-05:
Thanks for your response, I will close this issue.

closed 2019-02-05 by yogeshk19