ARMmbed/mbed-os #9516: mbed OS tools: Path for bootloader image is searched from config directory, not from project root

jarlamsa, 2019-01-28:
### Description
The path for bootloader image is searched from config directory, not from project root.
This causes an issue when program has multiple configuration files in a different folder than the project root, and the config file is issued in compilation with an --app-config flag.

It is happening here:
https://github.com/ARMmbed/mbed-os/blob/master/tools/config/__init__.py#L787

The right path would be project root instead of the config-file path.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jarlamsa, 2019-01-28:
@theotherjimmy could you please check?

theotherjimmy, 2019-01-28:
@jarlamsa This is the designed behavior.

> The right path would be project root instead of the config-file path.

I disagree, as this makes the paths in the config files confusing. Change your connfig files.

jarlamsa, 2019-01-29:
> this makes the paths in the config files confusing
How so, shouldn't the app-config config file be always referring to the app? 
So the paths in it should be referring to the root of the application, not root of the config file.

> Change your connfig files.
Since config files can recide in another directory than the root of the application, can you somehow refer to the root of the application in the path defined in the config file?

0xc0170, 2019-08-14:
@jarlamsa Is this issue still valid? Is there anything we should answer?

jarlamsa, 2019-08-14:
Need to check some point during this week. Don't close this yet.

teetak01, 2019-08-14:
This is still valid.

closed 2021-10-20 by ciarmcom