ARMmbed/mbed-os #9525: Flow control fails on STM32L476

cwq9999, 2019-01-28:
### Description

Flow control fails when the following commands are called in mbed-os 5.11.2 (a8f0c33e).

_serial.set_flow_control(SerialBase::Disabled);
 _serial.set_flow_control(SerialBase::RTSCTS, _rts, _cts); //  where tx = PB_10, rx = PB_11, rts = PB_14, cts = PB_13 

The error that is given is shown below:

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: obj_s->uart != (UARTName)NC
Location: 0x800CE7B
File: /extras/mbed-os.lib/targets/TARGET_STM/TARGET_STM32L4/serial_device.c+682
Error Value: 0x0
Current Thread: main  Id: 0x200026F0 Entry: 0x800D449 StackSize: 0x1000 StackMem: 0x200016F0 SP: 0x20002610 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0144
-- MbedOS Error Info 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-01-29:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2019-01-29:
Hi

I can reproduce the issue .

Thx for the report

jeromecoutant, 2019-01-29:
Hi
Please check #9534

jeromecoutant, 2019-01-29:
@0xc0170 
Could set "devices: st" label ?

PR #9534 (2019-02-04): STM32 Serial Flow Control
### Description

This fixes #9525 

Crash occurred when the flow control is set to disabled,
i.e. when serial_set_flow_control function was called with RTS and CTS values equal to "NC"

Correction is to check that used pin is using the same UART instance as TX and RX.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@cwq9999 @LMESTM
closed 2019-02-04 by 0xc0170