ARMmbed/mbed-os #9540: Ice Tea - configuration of allowed_platforms doesn't work as expected

maciejbocianski, 2019-01-30:
### Description
Icetea should allow to run the test with any of target listed on `allowed_platforms` config, but only first target on the list is recognized. (to run the test with specified target it have to be put on first place)
https://github.com/ARMmbed/mbed-os/blob/9030e5c6efbd9fe40baf3420394fc01eae8496b6/TEST_APPS/testcases/spi_com/test_spi_com.py#L68

@ARMmbed/mbed-os-test-team 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cmonr, 2019-01-30:
@ARMmbed/mbed-os-test-team

jupe, 2019-01-30:
it is already possible to select one platform from allowed list using argument `--platform_name`. See documentation here: 
https://github.com/ARMmbed/icetea/blob/master/doc/Icetea.md

without explicit platform_name selection icetea will pick first from allowed list.

I’m not sure if same option can be used with `mbed test`, probably not.

jonikula, 2019-01-31:
Also, if this is run through mbed test, the cli integration should handle setting the correct platforms in the correct way, so there should be no need to use the platform_name arg.

maciejbocianski, 2019-01-31:
Agree, the mbed cli should map `-m K64F` to `--platform_name` and pass it to icetea stuff

jonikula, 2019-01-31:
mbed cli already generates the suite it runs with Icetea in a manner where it should set the allowed_platforms to match the platform you are running with. If that does not happen then there's some issue there as well.

cmonr, 2019-01-31:
Hmm. Should this issue be moved into github.com/ARMmbed/mbed-os-tools?

ConradBraam, 2019-04-05:
I need to specify 2 (differing) targets, and when I use manually I find I have to edit the parameters every time. Using the 1st value breaks the separation of configuration data from code, so that driving multiple scenarios like I have, hard. Can we talk about a solution that also makes multiple "different" targets easy at the same time?

jonikula, 2019-04-05:
There might already be a way to do what you want to do, but it might not be readily apparent. Send me an email and we'll take a look.

0xc0170, 2019-08-14:
@maciejbocianski Have you found your answer ? Is there anything else for this issue or shall we close?

0xc0170, 2019-10-25:
I am closing as not relevant anymore, this won't be fixed.

closed 2019-10-25 by 0xc0170