ARMmbed/mbed-os #9558: DISCO_L072CZ_LRWAN1 loses TX bytes

juhhov, 2019-01-30:
### Description

 - What target does this relate to?
DISCO_L072CZ_LRWAN1
 - What toolchain (name + version) are you using?
ARM_GCC 6.3.1
  - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
mbed-cli 1.7.3
  - What is the SHA of Mbed OS (git log -n1 --oneline)?
a8f0c33eaa2c52babff9655417c36f4b5edd54d7
  - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)
Reproducable with mbed-client-cli:
1. Flash mbed-client-cli
2. Open screen
3. Paste "echo a66e745a-2494-11e9-a8ee-5f7b1e819461"
4. Expect same uuid to be echoed back
5. Actual: missing characters from echo or the uuid

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-01-30:
cc @ARMmbed/team-st-mcd @ARMmbed/mbed-os-wan

jeromecoutant, 2019-01-30:
Hi

> What is the SHA of Mbed OS (git log -n1 --oneline)?
> a8f0c33

Do you know if it was it OK with previous version ?

Thx

cmonr, 2019-01-30:
@juhhov Would you mind providing the exact output? Are they always the same missing characters?

juhhov, 2019-01-30:
@jeromecoutant I don't know. If you provide other version worth testing I can do it.

@cmonr They are not. Here is some rounds:
```
/>eho66e745a2494-11e9-ae-5f7b1e89461 
Command 'eho66e745a2494-11e9-ae-5f7b1e89461' not found.
/>eo a66e74a-2494-11e98ee-5f7b1819461 
Command 'eo' not found.
/>echa66e745-2494-11e98ee-5f7b1e19461 
Command 'echa66e745-2494-11e98ee-5f7b1e19461' not found.
/>eho a66e45a-2494-11-a8ee-5f7be819461 
Command 'eho' not found.
/>echa66e74a-2494-11e98ee-5f7b119461 
Command 'echa66e74a-2494-11e98ee-5f7b119461' not found.
/>eho a66e45a-2494-19-a8ee-5f71e819461 
Command 'eho' not found.
/>echa66e745-2494-11e98ee-5f7b1e19461 
Command 'echa66e745-2494-11e98ee-5f7b1e19461' not found.
/>echa66745a-2494-e9-a8ee-5fb1e819461 
Command 'echa66745a-2494-e9-a8ee-5fb1e819461' not found.
/>eco 6645a-249-11e9-a8ee-7b1e81946 
Command 'eco' not found.
/>ech66e745a-2-11e9-a8e5f7b1e8194 
Command 'ech66e745a-2-11e9-a8e5f7b1e8194' not found.
```

kjbracey-arm, 2019-01-31:
Which CLI test application are you using, exactly? I know some have been upgraded to use buffered `UARTSerial`, but I can see some using `RawSerial` just from a thread, which is going to be flakey.

How fast is the CPU running on this device? (You can print `SystemCoreClock`).

kjbracey-arm, 2019-01-31:
mbed-os-cliapp uses just `stdin` and `stdout`, and that should be fine, but `platform.stdio-buffered-serial` must remain on in its JSON configuration. Turning that off would break input.

kjbracey-arm, 2019-02-01:
Possibly related to #9577? Note that this is loss of serial characters on reception, not transmission, despite the title.

This test application is using `UARTSerial` via `platform.stdio-buffered-serial`, so an RX interrupt handler. #9577 is using `RawSerial` with a brute force foreground spin.

c1728p9, 2019-02-28:
Hi @juhhov what baudrate were you using when you saw the dropped bytes? And as @kjbracey-arm asked where you using UARTSerial or RawSerial?

If you were using a high baudrate, such as 115200, then running tickless of the the low power ticker will caused dropped bytes due to a uart overrun. If that is the cause then https://github.com/ARMmbed/mbed-os/pull/9785 should fix it.

juhhov, 2019-03-01:
Baudrate was 115200. As @kjbracey-arm mentioned ```"platform.stdio-buffered-serial": true``` was in use.

c1728p9, 2019-03-01:
@juhhov #9785 fixes a problems with interrupt latency which causes dropped bytes. There is a good chance this is the cause of your failure. Please try this change to see if it fixes your problem.

juhhov, 2019-03-04:
Built the same application against 5c24ffefa4b1c28d8f5aa9ec3e740538beaae436. The issue does not reproduce anymore. Thanks @c1728p9.

closed 2019-03-04 by juhhov