ARMmbed/mbed-os #9565: QSPIBlockDevice support for STM32F746ZG

RaEyE, 2019-01-31:
### Description

I noticed that QSPI and QSPIBlockDevice support has been added to MbedOS.
Therefore I wanted to try it out myself with an STM32F746ZG Nuleo board. The board supports, but is not listed within the QSPI mbed_lib.json files, altough there are STM32F4 and STM32F7 controllers present.

Is there a specific reason why there is no official support from mbed for STM32F746ZG controllers regarding QSPIBlockDevice?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-01-31:
Hi

Yes, QSPI is supported by STM32F746ZG.

You can then find the pins definition:
https://github.com/ARMmbed/mbed-os/blob/a12ab30f9802bd346aeb57c3b9b0ed676c587620/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F746xG/TARGET_NUCLEO_F746ZG/PeripheralPins.c#L384

But there is no embedded QSPI in the NUCLEO board, that's why it is not in the QSPI default file.

RaEyE, 2019-01-31:
Thank you very much, that explains it.
Then I will have to include / implement it myself since I want to integrate a Macronix MX25R QSPI Flash.

Now I at least know this implemention won't collide with a already existing one ;)

0xc0170, 2019-01-31:
Can this be closed as answered?

RaEyE, 2019-02-04:
Srry for the late reply.
Yes, please close it.

closed 2019-02-06 by TacoGrandeTX