ARMmbed/mbed-os #9574: NUCLEO_F767ZI: Ethernet TcpSocket::send is very slow

arjunak234, 2019-01-31:
### Description

> What toolchain (name + version) are you using?

GCC 7.1.0

> What is the SHA of Mbed OS (git log -n1 --oneline)?

mbed-os-5.11.2

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

With the following code, i barely get 1KiBps. But if add a `wait_ms(1)` after the send, speed increases to 1MiBps+

```
int main()
{

    EthernetInterface eth;
    eth.connect();

    TCPSocket socket;

    socket.open(&eth);
    socket.bind(80);
    socket.listen();

    static uint8_t buffer[1024];

    while (true) {
        nsapi_error_t err = 0;
        const auto s = socket.accept(&err);

        if (err == NSAPI_ERROR_OK) {
            nsapi_size_or_error_t result = sizeof(buffer);
            while (result == sizeof(buffer)) {
                result = s->send(buffer, sizeof(buffer));
            }
        }
        s->close();
    }
}

```

naveenkaje, 2019-02-01:
Hi Arjun,
Thanks for reporting this issue. Could you please give us more details on the hardware set up you measured this performance with?

Thanks!

arjunak234, 2019-02-01:
>Could you please give us more details on the hardware set up you measured this performance with

What do you mean by "hardware"? I'm using a Nucleo board, which is connected to my PC via a 100mbit router. On the PC i use 'nc' to establish a socket connection.

`nc 192.168.X.Y 80 > /dev/null`

cmonr, 2019-02-02:
> What do you mean by "hardware"?

@arjunak234 Wondows/Linux/Mac, OS versions, Python versions, pip module versions

SeppoTakalo, 2019-02-04:
1 Kib/s sounds a bit slow, as it will only call `send()` once a second.
1 Mib/s+ sound unrealistic, as the example code would call `send()` more than 1000 times a second, which cannot be possible when there is 1ms delay in between each call. It could be maximum of 1 Mib/s.

However, the LwIP stack is tuned to optimize memory usage, not speed. So it cannot achieve very high performance anyway...

SeppoTakalo, 2019-02-04:
@arjunak234 Can you try running TCPSOCKET_RECV_100K testcase on this platform?

I did check and those that we have in CI, can download 100 kiB in 3.5 - 3.7 seconds using buffer size of 1220.
So roughly 30 kiB/s.

I have not checked what is the speed that our test server limits...

0xc0170, 2019-08-14:
@arjunak234 Is this issue still valid, we haven't received any update since our latest responses here. This will be closed soon otherwise.

arjunak234, 2019-08-17:
>Can you try running TCPSOCKET_RECV_100K testcase on this platform 

All the tests in "mbed-os-tests-netsocket-tcp" pass. I can still reproduce it in 5.13.3

arjunak234, 2019-08-17:
I'm attaching a screenshot of packets captured by wireshark

[image]

adbridge, 2020-10-15:
We've updated our automation, I will fix the requirements .

mdkf, 2022-02-03:
Any update on this? I'm seeing a similar issue on the NUCLEO-F746ZG, NUCLEO-F429ZI.

0xc0170, 2022-02-07:
@ARMmbed/team-st-mcd has there been any similar reports ?

mdkf, 2022-02-09:
@0xc0170 I have also tested on the FRDM-K64F, the performance is even worse at 0.3 Mbps. I think this may not be hardware specific, but an issue with the lwIP implementation