ARMmbed/mbed-os #9577: Nucleo F411RE and F401RE dropping characters in serial communications when tickless is enabled.

offroadbis, 2019-01-31:
### Description

Characters are being dropped from serial communications at 115200 baud when tickless is enabled.  If I lower the baud rate to 57600 it seems to work.  If I remove the `"target.macros_add": ["MBED_TICKLESS","DEVICE_LPTICKER"],` line it works.  

This seems similar to the closed issue #8714 

Target Nucleo F411RE or Nucleo F401RE
Toolchain GCC_ARM
mbed-cli 
Mbed OS 5.11.2 a8f0c33eaa2c52babff9655417c36f4b5edd54d7

Nucleo 1 and Nucleo 2 can be either a F411RE or F401RE.

mbed_app.json
```
{
    "target_overrides": {
         "NUCLEO_F401RE": {
            "target.components_add": ["SD"],
            "target.macros_add": ["MBED_TICKLESS","DEVICE_LPTICKER"],
            "platform.stdio-baud-rate": 115200
        },
        "NUCLEO_F411RE": {
            "target.components_add": ["SD"],
        	"target.macros_add": ["MBED_TICKLESS","DEVICE_LPTICKER"],
            "platform.stdio-baud-rate": 115200
        }
    }
}
```

On the Nucleo 1 I put the following with the above mbed_app.json.
```
#include "mbed.h"
 
RawSerial Modem(PA_9, PA_10, 115200);
DigitalIn mybutton(USER_BUTTON);
DigitalOut led2(PA_5,0);

int main()
{
	LowPowerTimer t;
	char c;
	bool found = false;
	char respBuffer[1000];
	int testCompare=1;
    int bsize = sizeof(respBuffer);
    char expected[10] = "OK\r";
	int eLen = strlen(expected);
    Modem.set_flow_control(Serial::RTSCTS,PA_12,PA_11);
    printf("Ready...\n");
    t.start();
    while(1) {
		if(mybutton == 0) {
			respBuffer[0]=0;
			led2=1;
			t.reset();
			int i=0;
			while(t.read_ms() < 10000){
				while (Modem.readable()) {
					respBuffer[i] = Modem.getc();
					respBuffer[i+1]=0;
					i++;
				}
			}
			led2=0;
			printf("Found:%d\nOUTPUT:\n%s\nmilliSec: %d\n",found,respBuffer,t.read_ms());
		}
	}
}
```

On Nucleo 2 I put the following.
```
#include "mbed.h"
 
RawSerial Modem(PA_9, PA_10, 115200);
DigitalIn mybutton(USER_BUTTON);
DigitalOut led2(PA_5,0);

int main()
{
	Modem.set_flow_control(Serial::RTSCTS,PA_12,PA_11);
	char respBuffer[1000]="0123456789_0123456789\r0123456789_0123456789\r0123456789_0123456789\r0123456789_0123456789\r\n\r\nOK\r\n";
    Modem.set_flow_control(Serial::RTS,PA_12,PA_11);
    while(1) {
		if(mybutton == 0) {
			led2=1;
			printf("Sending...\n");
			Modem.puts(respBuffer);
			wait(1);
			led2=0;
		}
	}
}
```

Connect GND between devices
Connect Nucleo 1 PA_12 to Nucleo 2 PA_13 
Connect Nucleo 1 PA_13 to Nucleo 2 PA_12
Connect Nucleo 1 PA_10 to Nucleo 2 PA_9
Connect Nucleo 1 PA_9 to Nucleo 2 PA_10

Once both are running press the blue button to start listening on Nucleo 1.  Then press the blue button on Nucleo 2 to send the sample data.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

offroadbis, 2019-01-31:
The Modem set flow control lines in the code sample above should be commented out.  No characters are dropped with flow control CTS and RTS enabled.  `Modem.set_flow_control(Serial::RTSCTS,PA_12,PA_11);`

cmonr, 2019-01-31:
@ARMmbed/team-st-mcd

kjbracey-arm, 2019-02-01:
It does sound very similar to that other issue.   `Serial` and `RawSerial` are always going to be flakey for reception, but your test code is spinning hard enough using brute force that it should work. I think.

But my natural inclination seeing anyone reporting recception problems using unbuffered serial is to tell them not to attempt it. So...

Can you check using `UARTSerial` to get the actual data pump into interrupt space, so the application is reading from the interrupt-filled buffer?

LMESTM, 2019-02-01:
@kjbracey-arm  It takes about 2ms to exit the deep sleep mode on STM32. I don't understand how we can cope with this latency and still expect to receive all chars of Serial port. I think the Serial driver should prevent deep sleep entry when Rx interrupt is active - don't you think ? 
Then a more sophisticated 'protocol' can be implemented at device driver level, with a specific handshake mechanism for enabling / disabling deep sleep mode.
cc @jeromecoutant

jeromecoutant, 2019-02-01:
Note that "deep-sleep-latency" default value for STM32 is 3

kjbracey-arm, 2019-02-01:
I agree that it's impossible to wake from deep sleep using serial RX interrupt in time to process the character, but that will never currently be demanded from the HAL. 

Either code is actively spinning using 100% of CPU to check `serial_readable` (maybe implicitly inside the `serial_getc` implementation, or explicitly like here), or they will have attached an interrupt handler via `SerialBase::attach`, and that locks deep sleep.

So this does feel similar to #8714 - that was character loss due to the lpticker-based ticks happening while the CPU is active. (Tickless mode does "tick" while not idle).

LMESTM, 2019-02-01:
the issue title says : "when tickless is enabled"  .... and description explains that removing "MBED_TICKLESS" makes it work back again, so I was inclined to think the target actually sleeps ...  although the main thread seems to be spinning indeed. Needs to be tested

offroadbis, 2019-02-04:
When I posted the code initially I did not have the flow control lines removed. They were left over from my testing.  I meant to remove the flow control lines all together before posting the sample code. Once I made the initial post,  I could not figure out how to edit it to get rid of them, so I added a comment saying the flow control lines should be commented out.

LMESTM, 2019-02-04:
@offroadbis  there is a menu on the top right corner of your comments where you should be able to EDIT ...

LMESTM, 2019-02-04:
@offroadbis then we're back to my point, the system is less reactive when you move to TICKLESS and LP TICKER. There are many other things your application can do that will make your system slovwer. You'd better use interrupt based UART communication than a polling thread  ...

offroadbis, 2019-02-04:
Ah 3 little dots. Got it. Edited. Sorry for the confusion.  OK.  I will try to incorporate interrupts to read from the serial.  I just assumed that even when using TICKLESS and LP TICKER, if I go into a loop with no waiting or sleeping I would be able to read all incoming serial.  I therefore though it was a bug and thought I should report it.

LMESTM, 2019-02-04:
I think the scheduler will be less efficient in this mode, so you're maybe just above the limit without TICKLESS and LP_TICKER, but if you start adding more processing or other thread for your application, I'm afraid that it would result in the same loss of characters ... I think that BufferedSerial class already relies on interrupts.

offroadbis, 2019-02-04:
OK.  I will look into using BufferedSerial.  Thank all of you for your time and input.

closed 2019-02-04 by offroadbis

LMESTM, 2019-02-04:
@offroadbis my mistake, I think BufferedSerial does only bufferize for Tx, while you'd like an RX buffer here .

```
    /** Create a BufferedSerial port, connected to the specified transmit and receive pins
     *  @param tx Transmit pin
     *  @param rx Receive pin
     *  @param buf_size printf() buffer size
     *  @param tx_multiple amount of max printf() present in the internal ring buffer at one time
     *  @param name optional name
     *  @note Either tx or rx may be specified as NC if unused
     */
    BufferedSerial(PinName tx, PinName rx, uint32_t buf_size = 256, uint32_t tx_multiple = 4,const char* name=NULL);
```

so one of the various serial interrupt based examples should be better

kjbracey-arm, 2019-02-05:
`UARTSerial` is the in-tree class that provides buffering. (It is derived from `BufferedSerial`, but the name got changed just to avoid name conflicts).

Now, I'm not confident that if this test app fails (with no other threads running) that `UARTSerial` will work any better. Certainly you should be using `UARTSerial` normally to make reception reliable, but this specific test case was devoting 100% of foreground CPU time to reading the serial port - that should get the data out of the hardware as fast as an interrupt handler could.

(I believe the lptimer calls are made with IRQs fully masked, so serial IRQs get starved as much as the foreground thread).

> And I think there will always be cases where the test will fail because receiver CPU is busy doing something else ... Here is seems like it takes time reading the timer ... but anyway when your application will start doing something else, you'd lose characters anyway. So I assume you'd want to use HW flow control.

That's why you need an interrupt handler, and the interrupt latency in the system should be targeted at around the 100us mark which means you shouldn't lose non-flow-control UART characters.

An awful lot of boards do have no control on certain serial ports - particularly the debug terminal line to the USB, so as a quality-of-implementation aim we want it to be possible to reasonably reliably receive non-flow-controlled data via a buffered serial device like `UARTSerial`.

And it's kind of the canary in the coal-mine. The UART is the first thing to start visibly suffering with bad interrupt latency, but it will affect other devices too like radios.

offroadbis, 2019-02-05:
I changed my Nucleo 1 code to the following just to try and understand what is going on.  I still loose characters with the interrupt unless my main program waits for 0.1 seconds after the respBuffer changes size.  0.1 seconds is just a number I picked.  Not sure how low that value can go before loosing characters again.  If the main program doesn't ever wait, does that mean the interrupt will not get enough time to run, which is causing the character loss?  The reason I am looking into this is I am pretty sure the modem I am using doesn't properly use flow control because I loose characters even with it enabled.  That is why I am trying to get the serial to work as best as possible without flow control.  

```
#include "mbed.h"
 
RawSerial Modem(PA_9, PA_10,115200);
DigitalIn mybutton(USER_BUTTON);
DigitalOut led2(PA_5,0);
char respBuffer[1500] = {0};
volatile int bufferPosition = 0;
volatile int oldBufferPosition = 0;

void rx_interrupt(){
	if(bufferPosition>=1500){
		bufferPosition=0;
		oldBufferPosition=0;
	}
	while(Modem.readable()){
			respBuffer[bufferPosition++]=Modem.getc();
			respBuffer[bufferPosition]=0;
	}
}

int main()
{
	Modem.attach(callback(&rx_interrupt),Serial::RxIrq);
	printf("Ready...\n");
    while(1) {
    	if(bufferPosition!=oldBufferPosition){
    		wait(.1); // Characters are lost if this wait is commented out.
    		oldBufferPosition=bufferPosition;
    	}
		if(mybutton == 0) {
			wait(1);
			printf("%s\n",respBuffer);
		}
	}
}
```
No Character loss with wait(.1) statement included.

```
Ready...
0123456789_0123456789
0123456789_0123456789
0123456789_0123456789
0123456789_0123456789

OK
```

Character loss with wait(.1) commented out.

```
Ready...
0123456789_0123456789
0123456789_23456789
01456789_0123456789
0123456789_0123456789
```

Thank you for taking the time to help.

offroadbis, 2019-02-05:
I tried to change the Nucleo 1 code to use UARTSerial with the following code.

```
#include "mbed.h"
 
UARTSerial Modem(PA_9, PA_10,115200);
DigitalIn mybutton(USER_BUTTON);
char tempBuffer[256]="";
char respBuffer[1601]="";
bool done = false;

int main()
{
	int numread=0;
	printf("Ready...\n");
	while(!done){
		numread = Modem.read(tempBuffer,256);
		if(numread > 0){
			if((strlen(respBuffer)+numread) < 1600){
				strncat(respBuffer,tempBuffer,numread);
			}
			if (strstr(respBuffer,"OK") != NULL) {
				done=true;
			}
		}
	}
	printf("%s\n",respBuffer);
}
```

My output is as follows.  Still dropping characters.

```
0123456789_01234569
01256789_0123456789
01234789_01234567801234569_0123456789

OK
```
I only need to talk to a modem once a day.  Otherwise, the app is waiting.  It is a huge power saver to be able to use tickless.  We use 9-10mA instead of 16mA during the wait period when tickless is enabled.  Any other ideas?

kjbracey-arm, 2019-02-06:
And presumably putting the `wait` back in removes the loss again?

offroadbis, 2019-02-06:
I haven't found a place or length where the wait actually helps.  I almost always loose characters with the UARTSerial version, unless of course tickless is off.

LMESTM, 2019-02-07:
@kjbracey-arm @c1728p9 
should this ticket actually be re-opened or new one created.
If I understand well, we have an explanation today that it all comes down to scheduling disabling interrupts for a 80µs balck-out period - is that it ? 
Couldn't we improve or avoid this in a way or another ? 

Users are often basically interested at deep sleeping for a very long time, wake-up & do some SW handling, then back to deep sleep. For these use case, a RTC alarm API would  definitely help and could be a counter proposal or complement to TICKLESS ... the active period could be run using HR timer.

screamerbg, 2019-02-07:
CC @c1728p9

kjbracey-arm, 2019-02-07:
If the interrupt blackout period only happened as we were entering deep sleep, then it would be far less of an issue, or indeed no issue.  People relying on the interrupts would be locking the deep sleep anyway.

I guess the required interrupt latency for a no-FIFO 115200 serial port is 80us or so, and that really is a bit tight. It's very unfortunate that so many embedded devices are trying to run at that speed with no hardware FIFO - it is being somewhat optimistic about interrupt latency. A PC UART would have at least a 16-byte FIFO.

My understanding is that switching between SysTick for active and lpticker when deep sleeping caused  problems with clock synchronisation and drift, and using lpticker all the time was the solution. I don't know if it would be possible to overcome those issues.

Another possibility would be to not actually fully mask interrupts during the timer scheduling. I believe the timer scheduling here sets up IRQ priorities with the lptimer at "minimum, equal to SVC" for RTOS requirements. If that's true, then we may be able to just mask IRQs at that level, not totally, at least on v7M devices, letting the UART keep running.