ARMmbed/mbed-os #9578: Mbed-OS build fails for FUTURE_SEQUANA /family of targets with ARM Compiler 6.11

SenRamakri, 2019-01-31:
### Description

Mbed OS will be upgrading to ARMc6 Toolchain with upcoming releases. We are in the process of ensuring that all currently supported targets successfully compiles/passes tests with ARMC6 toolchain but we do have some targets currently failing the build with ARMc6 toolchain. The failures are in source code under targets folder and thus needs to be updated.

The error and warning details are as follow:
When compiling with ARMc6 toolchain for FUTURE_SEQUANA(and other related targets - FUTURE_SEQUANA_M0, FUTURE_SEQUANA_PSA, FUTURE_SEQUANA_M0_PSA) build fails with following errors:
```
[ERROR] In file included from ./targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8C63XX/TARGET_MCU_PSOC6_M4/system_psoc63_cm4.c:16:
In file included from ./targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8C63XX/device.h:30:
In file included from ./targets/TARGET_Cypress/TARGET_PSOC6/objects.h:27:
In file included from ./targets/TARGET_Cypress/TARGET_PSOC6/gpio_object.h:25:
In file included from ./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/gpio/cy_gpio.h:123:
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:457:12: warning: unknown attribute 'zero_init' ignored [-Wunknown-attributes]
    extern CY_NOINIT char_t cy_assertFileName[CY_MAX_FILE_NAME_SIZE];  /**< The assert buffer */
           ^
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:415:69: note: expanded from macro 'CY_NOINIT'
    #define CY_NOINIT           __attribute__ ((section(".noinit"), zero_init))
                                                                    ^
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:458:12: warning: unknown attribute 'zero_init' ignored [-Wunknown-attributes]
    extern CY_NOINIT uint32_t cy_assertLine;                           /**< The assert line value */
           ^
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:415:69: note: expanded from macro 'CY_NOINIT'
    #define CY_NOINIT           __attribute__ ((section(".noinit"), zero_init))
                                                                    ^
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:493:12: warning: unknown attribute 'zero_init' ignored [-Wunknown-attributes]
    extern CY_NOINIT cy_stc_fault_frame_t cy_faultFrame;    /**< Fault frame structure */
           ^
./targets/TARGET_Cypress/TARGET_PSOC6/device/drivers/peripheral/syslib/cy_syslib.h:415:69: note: expanded from macro 'CY_NOINIT'
    #define CY_NOINIT           __attribute__ ((section(".noinit"), zero_init))
                                                                    ^
./targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8C63XX/TARGET_MCU_PSOC6_M4/system_psoc63_cm4.c:413:7: error: expected '(' after 'asm'
__asm void Cy_MemorySymbols(void)
      ^
./targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8C63XX/TARGET_MCU_PSOC6_M4/system_psoc63_cm4.c:413:6: error: expected ';' after top-level asm block
__asm void Cy_MemorySymbols(void)
     ^
     ;
./targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8C63XX/TARGET_MCU_PSOC6_M4/system_psoc63_cm4.c:416:5: error: use of undeclared identifier 'EXPORT'
    EXPORT __cy_memory_0_start
    ^
3 warnings and 3 errors generated.
```
Targets Affected: FUTURE_SEQUANA, FUTURE_SEQUANA_M0, FUTURE_SEQUANA_M0_PSA, FUTURE_SEQUANA_PSA
Toolchain used: ARM Compiler 6.11 (Version 6.11)

Note that there may be other changes required to make build successful for ARMc6, so please address them as well. 

How to reproduce the issue?
1. Install ARMC6 compiler with version 6.11
2. Set ARMC6_PATH config variable as below.
mbed config --global ARMC6_PATH <Path to ARMC6 Toolchain bin directory>
3. Run the compile as: mbed compile -t ARMc6 -m <target name>

Note that you may see an "Error" when trying to use ARMc6 with "mbed compile". This is expected, but the compilation process should continue to proceed. We will be removing this Error/Warning once we switch to ARMc6 completely.

How to generate Pull Request to fix ARMc6 issues?
Please generate your pull requests addressing ARMc6 build issues with feature-armc6 as the target branch(not master branch).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

SenRamakri, 2019-01-31:
Tagging @ARMmbed/mbed-os-maintainers @ARMmbed/mbed-os-core @deepikabhavnani @kjbracey-arm @maclobdell 

@ARMmbed/mbed-os-maintainers @maclobdell - Is there a team label for Cypress team? I'm unable to find one. Can you please help find/add relevant Cypress people/team to this issue?

maclobdell, 2019-02-01:
cc @lrusinowicz

lrusinowicz, 2019-02-01:
What is the plan/deadline for merge of feature-armc6 to master, especially relative to PR #9481?
Fixes for this issue will likely result in conflicts with the above PR and I'd like to plan accordingly.

deepikabhavnani, 2019-02-01:
> What is the plan/deadline for merge of feature-armc6 to master

The last patch RC PR for 5.11 will be made on Feb 21st, PR for feature-armc6 merge to master will be created post that and CI will be switched to test armc6

@cmonr ^^^

cmonr, 2019-02-02:
@lrusinowicz The current plan (no one as argued against it at least) is what's here: https://github.com/ARMmbed/mbed-os/pull/9481#issuecomment-459513186

From what I can tell, #9481 is progressing well and should start testing early next week, and depending if any errors are found, might be brought in then as well. To hopefully expidite things, I kicked off testing on #9481 to get an idea what else is still needed.

Does that help?

SenRamakri, 2019-02-04:
@cmonr and @deepikabhavnani - Thanks for responding and providing the info.
@lrusinowicz - Hope that answers your question. Let us know if you need more info.

lrusinowicz, 2019-02-04:
Yes, thank you very much. I think I will wait with PR until #9481 is merged, since it will be easier for me to rebase than to risk conflicts during merge.

SenRamakri, 2019-02-22:
> Yes, thank you very much. I think I will wait with PR until #9481 is merged, since it will be easier for me to rebase than to risk conflicts during merge.

@lrusinowicz - https://github.com/ARMmbed/mbed-os/pull/9481 has been merged. Let us know if you need more info for fixing ARMC6 build failures.

lrusinowicz, 2019-03-04:
@SenRamakri 
@maclobdell 

Unfortunately, ARM Compiler 6.11 is not working for me at all. While trying to compile I get:
```
Compile [  0.3%]: except.S
Compile: C:/Keil_v5/ARM/ARMCLANG/bin\armasm --cpu=Cortex-M4.fp --cpreproc --cpreproc_opts=-c,--target=arm-arm-none-eabi,-mthumb,-Os,-Wno-armcc-pragma-push-pop,-Wno-armcc-pragma-anon-unions,-DMULADDC_CANNOT_USE_R7,-fdata-sections,-fno-exceptions,-MMD,-D_LIBCPP_EXTERN_TEMPLATE(...)=,-fshort-enums,-fshort-wchar,-DMBED_TRAP_ERRORS_ENABLED=1,-mcpu=cortex-m4,-mfpu=fpv4-sp-d16,-mfloat-abi=hard,-D__CORTEX_M4,-DCY8C6347BZI_BLD53,-DARM_MATH_CM4,-D__FPU_PRESENT=1,-D__MBED_CMSIS_RTOS_CM,-DMCU_PSOC6_M4,-D__CMSIS_RTOS,-DMBED_MPU_CUSTOM,-DPSOC6_DYNSRM_DISABLE=1,-DMBEDTLS_PSA_CRYPTO_C,-I.,-I.\mbed-os,-I.\mbed-os\cmsis,-I.\mbed-os\cmsis\TARGET_CORTEX_M,-I.\mbed-os\components,-I.\mbed-os\components\802.15.4_RF,-I.\mbed-os\components\802.15.4_RF\atmel-rf-driver,-I.\mbed-os\components\802.15.4_RF\atmel-rf-driver\atmel-rf-driver,-I.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source,-I.\mbed-os\components\802.15.4_RF\mcr20a-rf-driver,-I.\mbed-os\components\802.15.4_RF\mcr20a-rf-driver\mcr20a-rf-driver,-I.\mbed-os\components\802.15.4_RF\mcr20a-rf-driver\source,-I.\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver,-I.\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source,-I.\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\stm-s2lp-rf-driver,-I.\mbed-os\components\TARGET_PSA,-I.\mbed-os\components\TARGET_PSA\TARGET_MBED_SPM,-I.\mbed-os\components\TARGET_PSA\TARGET_MBED_SPM\COMPONENT_SPM_MAILBOX,-I.\mbed-os\components\TARGET_PSA\inc,-I.\mbed-os\components\TARGET_PSA\inc\psa,-I.\mbed-os\components\TARGET_PSA\services,-I.\mbed-os\components\TARGET_PSA\services\crypto,-I.\mbed-os\components\TARGET_PSA\services\crypto\COMPONENT_PSA_SRV_IPC,-I.\mbed-os\components\TARGET_PSA\services\platform,-I.\mbed-os\components\TARGET_PSA\services\storage,-I.\mbed-os\components\TARGET_PSA\services\storage\common,-I.\mbed-os\components\TARGET_PSA\services\storage\its,-I.\mbed-os\components\storage\blockdevice\COMPONENT_FLASHIAP,-I.\mbed-os\components\wifi,-I.\mbed-os\components\wifi\esp8266-driver,-I.\mbed-os\components\wifi\esp8266-driver\ESP8266,-I.\mbed-os\drivers,-I.\mbed-os\events,-I.\mbed-os\events\equeue,-I.\mbed-os\features,-I.\mbed-os\features\cellular,-I.\mbed-os\features\cellular\framework,-I.\mbed-os\features\cellular\framework\API,-I.\mbed-os\features\cellular\framework\AT,-I.\mbed-os\features\cellular\framework\common,-I.\mbed-os\features\cellular\framework\device,-I.\mbed-os\features\cellular\framework\targets,-I.\mbed-os\features\cellular\framework\targets\GEMALTO,-I.\mbed-os\features\cellular\framework\targets\GEMALTO\CINTERION,-I.\mbed-os\features\cellular\framework\targets\GENERIC,-I.\mbed-os\features\cellular\framework\targets\GENERIC\GENERIC_AT3GPP,-I.\mbed-os\features\cellular\framework\targets\MultiTech,-I.\mbed-os\features\cellular\framework\targets\MultiTech\DragonflyNano,-I.\mbed-os\features\cellular\framework\targets\MultiTech\DragonflyNano\PPP,-I.\mbed-os\features\cellular\framework\targets\QUECTEL,-I.\mbed-os\features\cellular\framework\targets\QUECTEL\BC95,-I.\mbed-os\features\cellular\framework\targets\QUECTEL\BG96,-I.\mbed-os\features\cellular\framework\targets\QUECTEL\M26,-I.\mbed-os\features\cellular\framework\targets\QUECTEL\UG96,-I.\mbed-os\features\cellular\framework\targets\TELIT,-I.\mbed-os\features\cellular\framework\targets\TELIT\HE910,-I.\mbed-os\features\cellular\framework\targets\UBLOX,-I.\mbed-os\features\cellular\framework\targets\UBLOX\AT,-I.\mbed-os\features\cellular\framework\targets\UBLOX\PPP,-I.\mbed-os\features\device_key,-I.\mbed-os\features\device_key\source,-I.\mbed-os\features\frameworks,-I.\mbed-os\features\frameworks\greentea-client,-I.\mbed-os\features\frameworks\greentea-client\greentea-client,-I.\mbed-os\features\frameworks\mbed-client-cli,-I.\mbed-os\features\frameworks\mbed-client-cli\mbed-client-cli,-I.\mbed-os\features\frameworks\mbed-client-randlib,-I.\mbed-os\features\frameworks\mbed-client-randlib\mbed-client-randlib,-I.\mbed-os\features\frameworks\mbed-client-randlib\mbed-client-randlib\platform,-I.\mbed-os\features\frameworks\mbed-coap,-I.\mbed-os\features\frameworks\mbed-coap\mbed-coap,-I.\mbed-os\features\frameworks\mbed-coap\source,-I.\mbed-os\features\frameworks\mbed-coap\source\include,-I.\mbed-os\features\frameworks\mbed-trace,-I.\mbed-os\features\frameworks\mbed-trace\mbed-trace,-I.\mbed-os\features\frameworks\nanostack-libservice,-I.\mbed-os\features\frameworks\nanostack-libservice\mbed-client-libservice,-I.\mbed-os\features\frameworks\nanostack-libservice\mbed-client-libservice\platform,-I.\mbed-os\features\frameworks\unity,-I.\mbed-os\features\frameworks\unity\unity,-I.\mbed-os\features\frameworks\utest,-I.\mbed-os\features\frameworks\utest\utest,-I.\mbed-os\features\lorawan,-I.\mbed-os\features\lorawan\lorastack,-I.\mbed-os\features\lorawan\lorastack\mac,-I.\mbed-os\features\lorawan\lorastack\phy,-I.\mbed-os\features\lorawan\system,-I.\mbed-os\features\lwipstack,-I.\mbed-os\features\lwipstack\lwip,-I.\mbed-os\features\lwipstack\lwip-sys,-I.\mbed-os\features\lwipstack\lwip-sys\arch,-I.\mbed-os\features\lwipstack\lwip\src,-I.\mbed-os\features\lwipstack\lwip\src\include,-I.\mbed-os\features\lwipstack\lwip\src\include\lwip,-I.\mbed-os\features\lwipstack\lwip\src\include\lwip\priv,-I.\mbed-os\features\lwipstack\lwip\src\include\lwip\prot,-I.\mbed-os\features\lwipstack\lwip\src\include\netif,-I.\mbed-os\feature [... cut ...]
  File "C:\testing\mbed-os\tools\toolchains\__init__.py", line 418, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\testing\mbed-os\tools\toolchains\__init__.py", line 481, in compile_queue
    raise ToolException(err)
ToolException: error: error opening 'C:\Users\LESZEK~1.d': permission denied
1 error generated.
Fatal error: A1905U: Pre-processor step failed for '.\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S'
1 Error, 0 Warnings

[mbed-9288] Working path "C:\testing" (program)
```
Has this compiler been tested / is it supporting Windows 7?

0xc0170, 2019-03-04:
yes, should work. 

Permission denied - it is trying to write to `C:\Users\LESZEK~1.d` but later working path is `C:/testing` ?

lrusinowicz, 2019-03-04:
The whole project is located in `C:\testing`. I have no idea where does `C:\Users\LESZEK~1.d` comes from as there is no reference in the assembler command line to anything in my home (user) directory and the presented path form gives no clues. Maybe some temporary file? But this is only a wild guess...

0xc0170, 2019-03-04:
@lrusinowicz What is the exact command ? I'll try to reproduce your path and command on Win 10 machine.

lrusinowicz, 2019-03-04:
@0xc0170 
I'm just trying to build tests:  `mbed test --compile -m FUTURE_SEQUANA -t ARMC6 -v`

0xc0170, 2019-03-04:
@ARMmbed/mbed-os-tools https://github.com/ARMmbed/mbed-os/issues/9578#issuecomment-469188493 - have you seen this error ? Can you help?

theotherjimmy, 2019-03-04:
@0xc0170 I have not seen that error. @bridadan Have you?

deepikabhavnani, 2019-03-04:
> Compile [  0.3%]: except.S
Compile: C:/Keil_v5/ARM/ARMCLANG/bin\armasm --cpu=Cortex-M4.fp --cpreproc

@lrusinowicz  - It looks the package used to build with ARMC6 is from Keil MDK installation, please use command `C:\Keil_v5\ARM\ARMCLANG\bin>armasm --help` to let us know which version of ARM compiler is used?

Output will be like this:
```
Product: MDK Professional 5.25 (Flex)
Component: ARM Compiler 6.9
```
Also please note we support 6.11 version of ARMC6 on master.

deepikabhavnani, 2019-03-04:
ARMC6 for sequana boards is disabled on master branch, till this issue is resolved, below change will be needed to reproduce the issue

```
diff --git a/targets/targets.json b/targets/targets.json
index 83340a7c46..b8781b787d 100644
--- a/targets/targets.json
+++ b/targets/targets.json
@@ -7855,7 +7855,7 @@
         "inherits": ["Target"],
         "macros": ["MBED_MPU_CUSTOM"],
         "default_toolchain": "GCC_ARM",
-        "supported_toolchains": ["GCC_ARM", "IAR", "ARMC5"],
+        "supported_toolchains": ["GCC_ARM", "IAR", "ARMC5", "ARMC6"],
         "core": "Cortex-M4F",
         "OUTPUT_EXT": "hex",
         "device_has": [

```

lrusinowicz, 2019-03-04:
@deepikabhavnani
Yes, I'm using Keil MDK 5.27 Prerelease. I've tried to use standalone 6.11 compiler, but its couldn't located proper license.
Please notice, that I seem to be getting exactly same issue while trying to use recent Mbed Studio Beta. That one was using its own instalation of 6.11 compiler.

deepikabhavnani, 2019-03-04:
@lrusinowicz - I got the `linker file not found error` on Mbed studio, but after changing targets.json as described here "https://github.com/ARMmbed/mbed-os/issues/9578#issuecomment-469358598", I was able to build for Future_sequana in debug mode.

lrusinowicz, 2019-03-05:
I've checked that the error occurs only when `-MMD` option is passed to the preprocessor from assembler. Without this option assembler works correctly.
So I would hypothesize, that this comes from how the preprocessor is invoked from the assembler on **Windows 7**. Likely, the option from which dependency filename is derived is not passed to the preprocessor resulting in an empty string and an ultimate filename like "$(HOME).d".

maciejbocianski, 2019-03-18:
@0xc0170 @lrusinowicz any progress with this ?
I'm getting the same error

```
$ mbed compile -m nucleo_f070rb -t ARM -v -c
Building project mbed-os-example-blinky (NUCLEO_F070RB, ARM)
Scan: mbed-os-example-blinky
Macros: -D__CORTEX_M0 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_ARMC6 -DTARGET_NAME=NUCLEO_F070RB -DDEVICE_USTICKER=1 -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM0 -DTARGET_M0 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_STM32F070RB -DDEVICE_SPI_ASYNCH=1 -DTARGET_NUCLEO_F070RB -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_LPTICKER=1 -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_LIKE_CORTEX_M0 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DMBED_BUILD_TIMESTAMP=1552941521.53
Compile [  0.1%]: mbed_tz_context.c
Compile: C:\Program Files\ARMCompiler6.11\bin\armclang -c --target=arm-arm-none-eabi -mthumb -Os -Wno-armcc-pragma-push-pop -Wno-armcc-pragma-anon-unions -DMULADDC_CANNOT_USE_R7 -fdata-sections -fno-exceptions -MMD -D_LIBCPP_EXTERN_TEMPLATE(...)= -fshort-enums -fshort-wchar -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m0 -D__ASSERT_MSG -std=gnu99 -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x20000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x4000 -D__CORTEX_M0 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_ARMC6 -DTARGET_NAME=NUCLEO_F070RB -DDEVICE_USTICKER=1 -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM0 -DTARGET_M0 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_STM32F070RB -DDEVICE_SPI_ASYNCH=1 -DTARGET_NUCLEO_F070RB -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_LPTICKER=1 -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_LIKE_CORTEX_M0 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DMBED_BUILD_TIMESTAMP=1552941521.53 @..\BUILD\NUCLEO_F070RB\ARM\.includes_8d98f860b1c35ca49a07f4d82a25b9aa.txt -include ..\BUILD\NUCLEO_F070RB\ARM\mbed_config.h -o ..\BUILD\NUCLEO_F070RB\ARM\mbed-os\cmsis\TARGET_CORTEX_M\mbed_tz_context.o ..\mbed-os\cmsis\TARGET_CORTEX_M\mbed_tz_context.c
[DEBUG] Return: 0
Compile [  0.2%]: mbed_fault_handler.c
Compile: C:\Program Files\ARMCompiler6.11\bin\armclang -c --target=arm-arm-none-eabi -mthumb -Os -Wno-armcc-pragma-push-pop -Wno-armcc-pragma-anon-unions -DMULADDC_CANNOT_USE_R7 -fdata-sections -fno-exceptions -MMD -D_LIBCPP_EXTERN_TEMPLATE(...)= -fshort-enums -fshort-wchar -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m0 -D__ASSERT_MSG -std=gnu99 -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x20000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x4000 -D__CORTEX_M0 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_ARMC6 -DTARGET_NAME=NUCLEO_F070RB -DDEVICE_USTICKER=1 -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM0 -DTARGET_M0 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_STM32F070RB -DDEVICE_SPI_ASYNCH=1 -DTARGET_NUCLEO_F070RB -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DDEVICE_LPTICKER=1 -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DCOMPONENT_NSPE=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_LIKE_CORTEX_M0 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD -DMBED_BUILD_TIMESTAMP=1552941521.53 @..\BUILD\NUCLEO_F070RB\ARM\.includes_8d98f860b1c35ca49a07f4d82a25b9aa.txt -include ..\BUILD\NUCLEO_F070RB\ARM\mbed_config.h -o ..\BUILD\NUCLEO_F070RB\ARM\mbed-os\cmsis\TARGET_CORTEX_M\mbed_fault_handler.o ..\mbed-os\cmsis\TARGET_CORTEX_M\mbed_fault_handler.c
[DEBUG] Return: 0
Compile [  0.4%]: except.S
Compile: C:\Program Files\ARMCompiler6.11\bin\armasm --cpu=Cortex-M0 --cpreproc --cpreproc_opts=-c,--target=arm-arm-none-eabi,-mthumb,-Os,-Wno-armcc-pragma-push-pop,-Wno-armcc-pragma-anon-unions,-DMULADDC_CANNOT_USE_R7,-fdata-sections,-fno-exceptions,-MMD,-D_LIBCPP_EXTERN_TEMPLATE(...)=,-fshort-enums,-fshort-wchar,-DMBED_TRAP_ERRORS_ENABLED=1,-mcpu=cortex-m0,-DMBED_ROM_START=0x8000000,-DMBED_ROM_SIZE=0x20000,-DMBED_RAM_START=0x20000000,-DMBED_RAM_SIZE=0x4000,-D__CORTEX_M0,-DTRANSACTION_QUEUE_SIZE_SPI=2,-DCMSIS_VECTAB_VIRTUAL,-DARM_MATH_CM0,-D__MBED_CMSIS_RTOS_CM,-DCMSIS_VECTAB_VIRTUAL_HEADER [... cut ...]
- mbed version: tag mbed-os-5.12.0-rc2
- tools version
$ pip list |grep mbed
mbed-cli           1.9.2
mbed-cloud-sdk     2.0.6
mbed-flasher       0.10.1
mbed-greentea      1.6.3
mbed-host-tests    1.5.3
mbed-ls            1.7.0
mbed-os-tools      0.0.4

- mbed config
[mbed] Global config:
GCC_ARM_PATH=C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin
ARM_PATH=C:\Keil_v5\ARM\ARMCC
IAR_PATH=C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.2\arm
ARMC6_PATH=C:\Program Files\ARMCompiler6.11\bin

cmonr, 2019-03-18:
@bridadan @maciejbocianski I'm wondering if https://github.com/ARMmbed/mbed-os/pull/10114 would happen to fix this issue.

lrusinowicz, 2019-03-19:
@maciejbocianski @cmonr @0xc0170 
I reviewed the changes in #10114 and I think they should fix the issue I'm having with compiler/assembler.

maciejbocianski, 2019-03-19:
I checked #10114 and it didn't help in my case

theotherjimmy, 2019-03-19:
I would not expect #10114 to resolve this issue.

-----------------

Notes: `C:\Users\macboc01.d` is not the file that the assembler should be looking for. Instead it should be `BUILD\NUCLEO_F070RB\ARM\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.d`. This might imply that the command line to the assembler is too long and being truncated incorrectly by windows.

lrusinowicz, 2019-03-20:
I'm sorry, I've mentally convoluted two different issues. Yes, #10114 will not resolve this.
@theotherjimmy @maciejbocianski 
On the other hand, I've been able to get compilation/assembly working on Win7 by modifying toolchain ARMC6 class `assemble` method to work like for generic class - preprocess .S first and then call assembler on a temporary file (instead of using `--cpreproc` option). So now I've unblocked possibility to fix Sequana issue.
Should I PR my tools patch?

maciejbocianski, 2019-03-20:
@lrusinowicz can you share this fix please

lrusinowicz, 2019-03-20:
@maciejbocianski 
Here https://github.com/lrusinowicz/mbed-os/commit/fa10eb89f1b943f0fef07889856c4f6a9722e592 is a temporary version of my fix. Definitely requires cleaning up, but I don't have time at this very moment.

theotherjimmy, 2019-03-20:
@Idanshenhav Yes please. Let's have that PR!

maciejbocianski, 2019-03-20:
In my case running terminal as admin 'solved'  the problem

PR #10346 (2019-04-09): FUTURE_SEQUANA: Add suport for ARMC6
### Description

This PR adds necessary set of changes to support ARMC6 compiler for FUTURE_SEQUANA family of targets. I've run default set of mbed tests on it and tests generally pass, except for a few failures related to device_key and flash API. I strongly suspect these are unrelated and would rather fix them separately.

Intended to fix #9578.

### Pull request type

    [ ] Fix
    [ ] Refactor
    [x] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-04-09 by cmonr