ARMmbed/mbed-os #9590: emac driver(smsc9220_emac.cpp) for CM3DS is not freeing buffers after sending packet - causing hang after sending large data

veerumgc, 2019-02-02:
### Description

### Issue request type
I  tried below logic which sends back data that it receives (512 bytes at a time).

char buffer[512];
char *p;
int r,remaining,size;

for(i=0;i<100;++i) 
{ 
     printf("Loop %d \n",i); 
     remaining = 0; 
     p = buffer; 
     while(remaining < 512) 
     { 
         r = socket.recv(p, 512); 
         if (r <= 0)
        {
             printf("Error! socket.recv() returned: %d\n", r); 
             break;
        } else { 
              remaining += r; 
              p += r;
        }
    }
   size= 512; 
   //Loop until whole request send 
   p = buffer; 
   while(size) 
   {
         r = socket.send(p, size);
         if (r < 0) { 
              printf("Error! socket.connect() returned: %d\n", r);
              break;
         }
         size -= r; 
         p+=r; 
   } 
}

Program hangs in loop 22. 

Following fix seems to solve the issue.

--- a/features/netsocket/emac-drivers/TARGET_ARM_SSG/COMPONENT_SMSC9220/smsc9220_emac.cpp
+++ b/features/netsocket/emac-drivers/TARGET_ARM_SSG/COMPONENT_SMSC9220/smsc9220_emac.cpp
@@ -168,6 +168,7 @@ bool SMSC9220_EMAC::link_out(emac_mem_buf_t *buf)
                                       true,
                                       (const char*)_memory_manager->get_ptr(buf),
                                       _memory_manager->get_len(buf));
+        _memory_manager->free(buf);
         if (error != SMSC9220_ERROR_NONE) {
             _TXLockMutex.unlock();
             return false;

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-08-14:
@veerumgc If not yet fixed, can you send PR fixing this?

0xc0170, 2019-08-14:
cc @ARMmbed/mbed-os-ipcore

closed 2019-09-11 by ciarmcom