ARMmbed/mbed-os #9597: Different compiler options for a specified files

mbedNoobNinja, 2019-02-02:
### Description
Is there any mechanism in the `mbed-CLI` system to build mbed app, in which a couple of files needs to be with different ARM compiler options (for example without "--gnu"), and the rest of the files just follows the options from the chosen profile.json ?

- How to pass these "different" files to the `mbed compile` command ?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-02-04:
@ARMmbed/mbed-os-tools team shall correct me, but this is not supported. 

We provide profiles and they are applied to entire codebase. Having specific rules for set of files would lead to more fragmentation. 

What is the use case to enable/disable specific flags for set of files?

mbedNoobNinja, 2019-02-04:
The case is x264 encoder library. It takes advantage of Neon ARM instructions. For best results and minimum encoding time & respectively higher FPS: The app **must** be compiled with "--arm", "--apcs=/hardfp", "--fpmode=fast", "--no_unaligned_access" & "--vectorize" and **no** "--gnu". I successfully make my own profile called x264.json and all this compiler flags are placed in the "common" section of the json file. So this flags are successfully applied to all files from the mbed app. The prolblem is that it can not be compiled. Some files from the network stack and platform folder specificly needs "--gnu", but all the rest doesn't. (because if "--gnu" is for all, the library works very slow at least **x2** times slower). I managed to accomplish this through the offline ARM ds5 IDE (I turn on "--gnu" just where it's need), but I want the app to be compilable through the CLI script too, because it's convenient (you don't need to put extra efforts to setup and fix the buggy exported project). The last one is real "pain" It took me averagely 1 day to fix already exported one.

kjbracey-arm, 2019-02-04:
Off on a slight tangent to the actual issue.

Does the `--gnu` flag really affect the compiler, in terms of optimisation? That seems surprising. Or is it tripping up some source code, fooling it into using less-optimal code because `__GNUC__` is defined?

If I'm not mistaken, `--apcs=/hardfp` should already be the default for ARM toolchain, if on a CPU with FP.

Does `--no_unaligned_access` really help? I would have thought that if code performs unaligned accesses, it's surely always better to use any CPU support for them, rather than to perform them synthetically, isn't it?

The `--fpmode=fast` and `--vectorize` really feel like they should be enable-able inside the code using pragmas, but it seems they're not. It would certainly be dubious to switch `fpmode` for other files, but you probably get away with it as there's very little "serious" FP in the OS itself.

mbedNoobNinja, 2019-02-04:
Most likely ` __GNUC__` fools the compiler and it uses less-optimal code.

bridadan, 2019-02-04:
@0xc0170, that's correct, you currently can't specify unique flags for specific files.

The only thing I can really think of would be compiling the files that you need with the special flags as a static library and then linking that into your application build.

cmonr, 2019-02-11:
@mbedNoobNinja Interesting idea.
Would you happen to know of any other build system that supports this kind of behavior?

mbedNoobNinja, 2019-02-12:
> Would you happen to know of any other build system that supports this kind of behavior?

As far as I know every IDE supports custom options for every file in the project, but as regards to the build systems the things are little bit tricky. I've heard that with Make you can do anything, but with the following ones I have no observations:

- Ant
- Scons
- CMake
- Jam

pan-, 2019-02-12:
> Would you happen to know of any other build system that supports this kind of behavior?

Any general (Make, Ant ...) build system can do that as programmers can create their own targets and rules for them. Build systems more specific to C/C++ can include such option out of the box (`set_source_files_properties` for CMake) and if they don't it is possible to create a different target with different options. Of course it is easier achieve more complex things when you expose a DSL to compose the build.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
Please reopen if you have further questions.