ARMmbed/mbed-os #9608: GCC_ARM build fails for v8m secure targets

mikisch81, 2019-02-05:
### Description

On ARM_Musca_A1 target (Cortex-M33), GCC_ARM build fails due to a premature use of the `cpu` attribute in the GCC class initializer:
```sh
[mbed] Exec "/usr/bin/python -u /home/micsch01/miki_dev/mbed_app/mbed-os/tools/make.py -t GCC_ARM -m ARM_MUSCA_A1_S --source ./mbed-os/components/TARGET_PSA --source ./mbed-os/drivers --source ./mbed-os/platform --source ./mbed-os/cmsis --source ./mbed-os/hal --source ./mbed-os/targets --source ./mbed-os/features/storage/blockdevice --source ./mbed-os/components/storage/blockdevice --source ./mbed-os/features/storage/kvstore/tdbstore --source ./mbed-os/features/storage/kvstore/include --source ./mbed-os/features/storage/kvstore/conf/tdb_internal --source ./mbed-os/features/mbedtls --build ./BUILD/ARM_MUSCA_A1_S/GCC_ARM -v -DNO_GREENTEA" in "/home/micsch01/miki_dev/mbed_app"
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
Traceback (most recent call last):
  File "/home/micsch01/miki_dev/mbed_app/mbed-os/tools/make.py", line 73, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "/home/micsch01/miki_dev/mbed_app/mbed-os/tools/build_api.py", line 512, in build_project
    app_config=app_config, build_profile=build_profile, ignore=ignore)
  File "/home/micsch01/miki_dev/mbed_app/mbed-os/tools/build_api.py", line 342, in prepare_toolchain
    target, notify, macros, build_dir=build_dir, build_profile=profile)
  File "/home/micsch01/miki_dev/mbed_app/mbed-os/tools/toolchains/gcc.py", line 63, in __init__
    self.cpu.append("-mcmse")
AttributeError: 'GCC_ARM' object has no attribute 'cpu'
[mbed] ERROR: "/usr/bin/python" returned error.
```

The use of `cpu` is here: [https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L63](https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L63) .

The 1st time `cpu` is set is here: [https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L77-L80](https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L77-L80).

Note: CI did not detect it as it only happens for V8M secure targets which is currently built in an external repository.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

CC @ARMmbed/mbed-os-tools @deepikabhavnani

0xc0170, 2019-02-05:
> Note: CI did not detect it as it only happens for V8M secure targets which is currently built in an external repository.

Should this be tested ? If not now, when? is there a ticket for this addition to CI?

Please provide steps how to reproduce (if this is internal repository, via email is also fine). Just to have this fixed today.

0xc0170, 2019-02-05:
@mikisch81 If this affects you, I would move line 77

```
--- a/tools/toolchains/gcc.py
+++ b/tools/toolchains/gcc.py
@@ -54,6 +54,10 @@ class GCC(mbedToolchain):
             self.flags["ld"].append("--specs=nano.specs")

         core = target.core
+        if core == "Cortex-M33":
+            self.cpu = ["-march=armv8-m.main"]
+        else:
+            self.cpu = ["-mcpu={}".format(cpu.lower())]
         if CORE_ARCH[target.core] == 8:
             # Add linking time preprocessor macro DOMAIN_NS
             if target.core.endswith("-NS"):
@@ -74,10 +78,6 @@ class GCC(mbedToolchain):
             "Cortex-M33F": "cortex-m33+no_dsp",
             "Cortex-M33FE": "cortex-m33"}.get(core, core)

-        if core == "Cortex-M33":
-            self.cpu = ["-march=armv8-m.main"]
-        else:
-            self.cpu = ["-mcpu={}".format(cpu.lower())]
```

I can send PR proposal for this

mikisch81, 2019-02-05:
@0xc0170 this proposal would break Cortex-M33-NS targets, because `core` will be set to `Cortex-M33-NS` and then in the `if` statement which you moved up it will go to the `else` and set `self.cpu` to a wrong value.

I have a workaround which works for me which is to set `self.cpu = []` in the begining and just append afterwards what is needed.

I can wait for @deepikabhavnani or @ARMmbed/mbed-os-tools for the proper solution.

mikisch81, 2019-02-05:
> Should this be tested ? If not now, when? is there a ticket for this addition to CI?
>
> Please provide steps how to reproduce (if this is internal repository, via email is also fine). Just to have [...]

@0xc0170 I think we should hear what @ARMmbed/mbed-os-tools say about testing this flow.
To reproduce it you will need to build a secure v8m target which can only be built by cloning https://github.com/ARMmbed/mbed-os-example-secure-app.git.

deepikabhavnani, 2019-02-05:
> I have a workaround which works for me which is to set self.cpu = [] in the begining and just append afterwards what is needed.

Creating self.cpu in advance and performing append for mcmse and cpu attributes seems good.

0xc0170, 2019-02-05:
@deepikabhavnani Can you create PR please?

mikisch81, 2019-02-05:
@0xc0170 I am opening right now, @deepikabhavnani did you already open?

theotherjimmy, 2019-02-05:
@mikisch81 I'm happy with creating `self.cpu` in advance. Feel free to submit a PR to that affect.

mikisch81, 2019-02-05:
> 
> 
> @mikisch81 I'm happy with creating `self.cpu` in advance. Feel free to submit a PR to that affect.

@theotherjimmy #9614

mikisch81, 2019-02-10:
Fixed by #9614

closed 2019-02-10 by mikisch81