ARMmbed/mbed-os #9620: FUTURE_SEQUANA: failing to pass block device tests.

ithinuel, 2019-02-05:
### Description

`mbed test -t GCC_ARM -m FUTURE_SEQUANA -n mbed-os-features-storage-tests-blockdevice-general_block_device -v`  fails with 

```
[1549386440.54][CONN][RXD] >>> Running 8 test cases...
[1549386440.54][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing Init block device}}, queued...
[1549386440.54][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing read write random blocks}}, queued...
[1549386440.55][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing multi threads erase program read}}, queued...
[1549386440.55][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing contiguous erase, write and read}}, queued...
[1549386440.57][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing BlockDevice erase functionality}}, queued...
[1549386440.57][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing program read small data sizes}}, queued...
[1549386440.58][CONN][RXD]
[1549386440.58][CONN][RXD] >>> Running case #1: 'SD Testing Init block device'...
[1549386440.58][CONN][RXD]
[1549386440.58][CONN][INF] found KV pair in stream: {{__testcase_name;SD Testing Deinit block device}}, queued...
[1549386440.58][CONN][INF] found KV pair in stream: {{__testcase_name;DEFAULT Testing get type functionality}}, queued...
[1549386440.59][CONN][INF] found KV pair in stream: {{__testcase_start;SD Testing Init block device}}, queued...
[1549386440.60][CONN][RXD] Test Init block device.
[1549386440.60][CONN][RXD]
[1549386440.60][CONN][RXD]
[1549386440.60][CONN][RXD] ++ MbedOS Error Info ++
[1549386440.60][CONN][RXD] Error Status: 0x80FF0100 Code: 256 Module: 255
[1549386440.60][CONN][RXD] Error Message: Fatal Run-time error
[1549386440.60][CONN][RXD] Location: 0x1008AE07
[1549386440.60][CONN][RXD] Error Value: 0x0
[1549386440.61][CONN][RXD] Current Thread: main  Id: 0x8011B28 Entry: 0x10086BDD StackSize: 0x1770 StackMem: 0x80127D8 SP: 0x8013CF4
[1549386440.63][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=FUTURE_SEQUANA
[1549386440.63][CONN][RXD] -- MbedOS Error Info --
```
Using mbed's cishield and this configuration :

```json
"target.components_add"             : ["SD"],
"sd.SPI_MOSI"                       : "D11",
"sd.SPI_MISO"                       : "D12",
"sd.SPI_CLK"                        : "D13",
"sd.SPI_CS"                         : "D10",
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

MarceloSalazar, 2019-02-05:
FYI @lrusinowicz
Let me know if you need more information to reproduce this

lrusinowicz, 2019-02-06:
> Using mbed's cishield

Which one is that?

> "sd.SPI_CS"                         : "D10",

How does SD component use this signal? Does it try to configure it as hardware-controlled part of SPI object? This won't work at the moment; current HAL implementation supports only a single SS signal per interface and that's P9_3, available only on microBUS connector.
For other SS signals one has to resort to GPIO - I think #9469 could help here.
Trying to configure hardware SS signal on D10 will result in Mbed error generated in a HAL layer - likely reason of this failure.

ithinuel, 2019-02-06:
The [SD driver](https://github.com/ARMmbed/mbed-os/blob/master/components/storage/blockdevice/COMPONENT_SD/SDBlockDevice.cpp#L254) control the CS as a GPIO and do not pass it to the SPI driver.

The ci-test-shield is : https://github.com/ARMmbed/mbed-HDK-Eagle-Projects
But any micro sd adapter would do the trick, for example :
- https://www.mikroe.com/microsd-click
- https://www.amazon.co.uk/dp/B07KF9C34X

You may have to pay attention to the voltages supported by the host board and the µsd adapter.

cmonr, 2019-02-11:
@ithinuel Does https://github.com/ARMmbed/mbed-os/pull/9647 resolve your issue?

0xc0170, 2019-02-20:
> @ithinuel Does #9647 resolve your issue?

Should be, reopen if not

closed 2019-02-20 by 0xc0170