ARMmbed/mbed-os #9623: LittleFS and dead cells / sectors in flash

Alex-EEE, 2019-02-05:
### Description

Does LittleFS ( or any of the BlockDevice drivers it's built on ) handle dead cells/sectors in Flash?  I know it does wear leveling, but what happens when writes fail to a given address?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-02-11:
@ARMmbed/mbed-os-storage @geky

geky, 2019-02-12:
Hi @Alex-EEE, LittleFS detects all write failures. It does this by reading any written data back immediately after writing to verify that the write was successful. If a write fails, LittleFS allocates a new block and relocates the data that was destined for the bad block.

For file data this is relatively easy, the data structure is copy-on-write, so we can switch to a new block without touching anything else. For metadata this is a bit harder since we have to worry about power-loss as we update the filesystem. The corner cases are documented in the [DESIGN.md](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/littlefs/DESIGN.md) file if you're interested.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@Alex-EEE Please reopen if you have further questions.

Alex-EEE, 2019-04-30:
Geky's answer was sufficient, thanks!