ARMmbed/mbed-os #9626: mbed_rtx.h file cleanup

jeromecoutant, 2019-02-06:
### Description

Proposition is to clean every targets/TARGET_xxx/mbed_rtx.h files

- INITIAL_SP definition is no more used (since #9092)
- HEAP_START/HEAP_SIZE/ISR_STACK_START/ISR_STACK_SIZE seems to be common for lots of targets ?

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

0xc0170, 2019-02-07:
cc @mprse

mprse, 2019-02-07:
@deepikabhavnani prepares the general solution for moving to 2 region memory model. Handling of `INITIAL_SP` and `HEAP_START/HEAP_SIZE/ISR_STACK_START/ISR_STACK_SIZE` macros is a part of this process.

deepikabhavnani, 2019-02-07:
@jeromecoutant 
Agree with this enhancement, INITIAL_SP is not used in Mbed OS 5
https://github.com/ARMmbed/mbed-os/blob/75461864589694ed08fa52354f4e40cf00471f29/rtos/TARGET_CORTEX/mbed_boot.c#L46

But we cannot remove it to maintain backward compatibility with Mbed 2 + CMSIS RTOS version 2. We will pick up this enhancement in future versions. 

Along with this cleanup of linker files is needed as well. We have duplicate exports like `__heap_limit` and `__HeapLimit`

cmonr, 2019-02-11:
@SenRamakri Fyi

closed 2019-10-31 by ciarmcom