ARMmbed/mbed-os #9627: ESP8266EX doesn't work when IO15 (UART0_RTS) isn't pulled down

soramame21, 2019-02-06:
### Description
EESP8266EX doesn't answer "AT" commaned at following API inside components\wifi\esp8266-driver\ESP8266Interface.cpp when IO15 pin isn't pulled down.
`ESP8266Interface::_init(void)  >> _esp.at_available()`

The repository I'm using :  https://github.com/ARMmbed/pelion-enablement
Compiling / running command :
```
mbed test -t ARM -m K64F -n simple-mbed-cloud-client-tests* --compile -v
mbed test -t ARM -m K64F -n SIMPLE-MBED-CLOUD-CLIENT-TESTS-BASIC-NET-SINGLE --run -v
```
Configuration at mbed_app.json :
```
            "target.network-default-interface-type" : "WIFI",
            "target.components_add"             : ["ESP8266"],
            "esp8266.debug"                     : false,
            "esp8266.rx"                        : "D0",
            "esp8266.tx"                        : "D1",
            "esp8266.cts"                       : null,
            "esp8266.rts"                       : null,
            "esp8266.rst"                       : "D7",
            "esp8266.provide-default"           : true,
```
ESP-WROOM-02 Wi-Fi shield information :
https://www.switch-science.com/catalog/2619/
https://s3-ap-northeast-1.amazonaws.com/switch-science.public/schematic/ESP-WROOM-02_Shield/ESP_WROOM_02_shield_rev3.pdf

ESP8266EX Firmware version : v1.7
Hardware patch I made on the shield :
I intend to enable hardware flow control feature connecting this rts to K64F cts at D2.
> **Romoved R9 (10K Ohm) resister pulling down IO15 (RTS pin).**

then ESP8266EX become not answering "AT" command sent through K64F tx (D1) anymore. 
once I pull down by connecting a 10k Ohm resister at middle of IO15 and GND, ESP8266EX starts responding well.

## the point
Can ESP8266 IO15 be used as RTS for hardware flow control instead of pulling down?

## the charts
The **not working** signal chart with no pulling down ESP8266 IO15 :
Signal 1 : K64F Rx 
Signal 2 : K64F Tx
Signal 3 : ESP8266 IO13 CTS
Signal 4 : ESP8266 IO15 RTS

1. ESP8266 driver give a reset signal from K64F D7 pin. 
2. **ESP8266 IO15 RTS is set High for over 20ms, then turned to Low. this is controlled by ESP8266EX, not K64F at all!** 
3. "AT" command is sent from K64F Tx pin, but no answer.
  
[image]

The **working** signal chart with pulling down ESP8266 IO15 :
 
1. ESP8266 driver give a reset signal from K64F D7 pin. 
2. **ESP8266 IO15 RTS cannot set High because of the pulling down.** 
3. "AT" command is sent from K64F Tx pin, respond well from ESP8266.

[image]

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

cmonr, 2019-02-11:
@ARMmbed/mbed-os-ipcore

SeppoTakalo, 2019-02-12:
Hi @soramame21 

I fail to see how this is relevant to our software.

If the device stopped working after you unsoldered a component from it, the obvious fix would be to solder it back.

Is there any evidence the it needs to be removed in order to use the RTS pin?

soramame21, 2019-02-12:
Hi @SeppoTakalo, cc @marcuschangarm

SeppoTakalo, 2019-02-12:
We use the `Arduino ESP8266 WiFi Shiald Version 1.0 by WangTongze`

https://www.banggood.com/ESP8266-ESP-12E-UART-WIFI-Wireless-Shield-TTL-Converter-For-Arduino-UNO-R3-Mega-p-1094091.html?akmClientCountry=GB&aff_key=21k3_ait23&utm_source=ShareASale&utm_medium=aff&sscid=21k3_ait23&cur_warehouse=CN

https://elfquake.wordpress.com/2018/02/25/arduino-esp8266-shiald-step-by-step/

HW flow control is working verified to work with this one and it seems to have the pull-down, at least according to one of those links.

soramame21, 2019-02-12:
@SeppoTakalo Thanks for sending the links. but at the linked page, I can see GPIO15 pin is pulled down instead of being used as RTS pin, **which is exactly as the same as this one**. 
Does the module is working with hardware flow control?
[image]

SeppoTakalo, 2019-02-12:
Yes it works.

Just route the RTS pin to K64F.

soramame21, 2019-02-12:
Which one is the RTS pin? IO15?

SeppoTakalo, 2019-02-13:
Pinouts are defined in datasheets.
I cannot know which pinout your module uses.

Your image refers that on ESP side it is GPIO15.

On K64F and K66F we have verified following pinouts to work:
```
        "K64F": {
            "esp8266.provide-default": true,
            "esp8266.tx": "PTC4",
            "esp8266.rx": "PTC3",
            "esp8266.rts": "PTC1",
            "esp8266.cts": "PTC2",
            "esp8266.rst": "D8"
        },
        "K66F": {
            "esp8266.provide-default": true,
            "esp8266.tx": "PTC4",
            "esp8266.rx": "PTC3",
            "esp8266.rts": "PTC1",
            "esp8266.cts": "PTC2",
            "esp8266.rst": "D8"
        },
```

soramame21, 2019-02-13:
@SeppoTakalo thanks for the information. the issue I filed (=I tested) is the IO15 pin on ESP module cannot be used for RTS. once IO15 is not pulled down, ESP module doesn't answer 1st contact ("AT" commad) from K64F. I'd like to know how to make IO15 working without pulling down.

cmonr, 2019-02-13:
@ARMmbed/device-os-custom-engineering-and-support Mind taking over?

soramame21, 2019-02-14:
@cmonr why you reject to answer my issue??

cmonr, 2019-02-14:
@SeppoTakalo closed the issue internally because it wasn't listed as a bug. 
> I cannot know which pinout your module uses.

When I made the comment, I reopened the internal ticket, whcih _should_ have reopened this as well, but that seems to have failed. I'll do it manually for now.

soramame21, 2019-02-14:
@cmonr @SeppoTakalo I didn't get an expected answer for judging if this is a bug. I need the explanation from a ESP driver expert.

SeppoTakalo, 2019-02-14:
@soramame21 
Do not remove any components.

Then re-test.

I don't know why do you think that you need to de-solder some of the components on the board. Just don't do it.

Use unmodified board, and it works.

I have just sent you the pinout that we have verified to work.

If you remove the pull-down, the ESP never answers, and the driver is never able to give command that enables the HW flow control. Please note that both ESP8266 and our driver boots up in a mode where flow control is not enabled. After the boot, we need first `AT\r\n` commands to pass, before we can issue commands that enable the flow control on ESP side. Only after this first handshake, the flow control is enabled on device side.

soramame21, 2019-02-17:
@SeppoTakalo Thanks for your reply. I understood your point is IO15 on ESP8266 can be used as RTS pin for hardware flow control, although the IO15 pin is pulled down. the resister (R2) soldered on the shield originally should not be removed. I'll test on an ESP8266 shield without removing R2 resister connecting to IO15 pin. I'll enable hardware flow control configuration at mbed_app.json and connect CTS pin on K64F to IO15 (RTS pin) on ESP8266.

soramame21, 2019-03-01:
@SeppoTakalo cc @marcuschangarm I had verified following items.

1. IO15 must be pulled down. It is the nature of ESP8266.
2. the Pulled down pin is used as RTS pin of ESP8266.
3. ESP8266 with hardware flow control can with K64F in 2 modes.

```
RTSCTS mode
RTS (K64F, PTC18) - CTS (IO13, ESP8266)
CTS (K64F, D2) - RTS (IO15, ESP8266)
mbed_app.json
            "esp8266.rx"                        : "D0",
            "esp8266.tx"                        : "D1",
            "esp8266.cts"                       : "D2",
            "esp8266.rts"                       : "PTC18",
```
```
CTS mode
RTS (K64F, PTC18) - CTS (IO13, ESP8266)
mbed_app.json
            "esp8266.rx"                        : "D0",
            "esp8266.tx"                        : "D1",
            "esp8266.cts"                       : null,
            "esp8266.rts"                       : "PTC18",
```

I close this issue now.
Thank you for your patience!

closed 2019-03-01 by soramame21