ARMmbed/mbed-os #9629: Wrong AnalogIn values on NUCLEO_L476RG target

sillevl, 2019-02-06:
### Description

When using AnalogIn to measure voltages on A0 (and other pins) on NUCLEO_L476RG results in wrong values. It is hard to say, but it may be that the resulting value is offset by 0.5, and is divided by half.

When applying 0V to the A0 pin, the measured value is around 0.486..., When applying 3.3V to the pin A0, the measured value is arount 0.989...

Tested the same code on NUCLEO_L073RZ and NUCLEO_F767ZI without any problems.

#### Target

`NUCLEO_L476RG`

#### Toolchain

`GCC_ARM`

#### Tools

`mbed-cli 1.8.3`

#### Mbed OS version

`57d8915a899a59999d7e2a9695e20a8a18cb54bd`

#### Steps to reproduce

Running the code found at:
https://os.mbed.com/teams/mbed/code/AnalogIn-HelloWorld/docs/0a8f602aeec9/main_8cpp_source.html

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jeromecoutant, 2019-02-06:
Hi

If you replace ADC_SAMPLETIME_47CYCLES_5 by ADC_SAMPLETIME_92CYCLES_5

https://github.com/ARMmbed/mbed-os/blob/57d8915a899a59999d7e2a9695e20a8a18cb54bd/targets/TARGET_STM/TARGET_STM32L4/analogin_device.c#L120

Is it better ?

sillevl, 2019-02-06:
@jeromecoutant changing the sConfig.SamplingTime does not solve this. The result is exactly the same as before.

sillevl, 2019-02-09:
@jeromecoutant anything other that I could try or should investigate?

pcordemans, 2019-02-14:
I've tried the example in the online mbed compiler and the issue did not occur.

fbobrow, 2019-06-21:
I have the exactly same problem as @sillevl described. Have you figure out yet?

I have tried with NUCLEO-L432KC and DISCO-L475VG-IOT01A, both yielding the same problem. With online Mbed Compiler this doesn't happens (just as @pcordemans described). However, the online compiler utilizes Mbed 2.0 and not Mbed OS 5.0.

My hypothesis is that there is a bug with AnalogIn in Mbed OS 5.0.

LMESTM, 2019-06-21:
@ARMmbed/team-st-mcd

jeromecoutant, 2019-07-19:
Hi
I think issue should be closed after #11072 
Maybe someone could check ?
Thx

closed 2019-08-14 by 0xc0170