ARMmbed/mbed-os #9632: NRF_LOG_* -> Compile errors

wlamprecht, 2019-02-06:
### Description

Hi

I am trying to enable the NRF_LOG by setting NRF_LOG_ENABLED to 1 in file sdk_config.h 
But thee needed resources are not included. 
Error message:
nrf_fprintf.h: No such file or directory

Are the needed resources available somewhere? 

Using:
Mbed 5.11
NRF52840 
Target SDK 14_2

Thanks, 
Werner 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

pan-, 2019-02-07:
@wlamprecht The missing files are in the Nordic SDK14.2 (`Target SDK 14_2`) that you can find on  Nordic's website.

wlamprecht, 2019-02-07:
@pan- I included the missing files from the nordic website already. But the debug log does not work with them. 
I get errors like: 
mbed-os/external/mbedtls/include/mbedtls/check_config.h:56:2: #error "The NET and TIMING modules are not available for mbed OS - please use the network and timing functions provided by mbed OS"
 
Do I put the files in the wrong place or does it not work in combination with mbed?

0xc0170, 2019-02-07:
> Do I put the files in the wrong place or does it not work in combination with mbed?

It looks like the combination is not supported.

```
#if defined(TARGET_LIKE_MBED) && \
    ( defined(MBEDTLS_NET_C) || defined(MBEDTLS_TIMING_C) )
```

Seems like that the new files are causing this to be triggered

wlamprecht, 2019-02-08:
Do you have any suggestions, on how to solve this problem?

cmonr, 2019-02-11:
@TacoGrandeTX @elm3 Thoughts?

TacoGrandeTX, 2019-02-13:
@wlamprecht Have you made any additional mods to sdk_config.h?   On v5.11.2 the following changes worked for me:  
```
 #ifndef NRF_FPRINTF_ENABLED
-#define NRF_FPRINTF_ENABLED 0
+#define NRF_FPRINTF_ENABLED 1
 #endif

 #ifndef NRF_STRERROR_ENABLED
-#define NRF_STRERROR_ENABLED 0
+#define NRF_STRERROR_ENABLED 1
 #endif

 #ifndef NRF_LOG_ENABLED
-#define NRF_LOG_ENABLED 0
+#define NRF_LOG_ENABLED 1
 #endif

 #ifndef NRF_LOG_CLI_CMDS
-#define NRF_LOG_CLI_CMDS 1
+#define NRF_LOG_CLI_CMDS 0
 #endif
```

I disabled NRF_LOG_CLI_CMDS simply because it required more files to be brought in and I didn't think it was needed.  

As for files added from the v14.2 SDK, I copied the ```external/fprintf``` folder to ```targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries/fprintf/```

I haven't actually adding logging to a program, but a build with those definitions is successful.  Perhaps you are seeing the error only after adding the logging calls (i.e.,  NRF_LOG_INIT() )?

TacoGrandeTX, 2019-03-08:
Closing... feel free to re-open if you still have a problem.

wlamprecht, 2019-03-20:
Hi @TacoGrandeTX, 
No, I have no additional changes in the file sdk_config.h. I'm using Mbed-OS 5.11.5

I adapted sdk_config.h according to your suggestions

Downloaded the folder fprintf from device SDK 14_2 and copied to path targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries

I get following linker error: 

tools.utils.ToolException: b"(...)/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries/experimental_log/src/nrf_log_frontend.o: In function `nrf_log_backend_put':
(...)/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/libraries/experimental_log/nrf_log_backend_interface.h:172: undefined reference to `__start_log_dynamic_data'\ncollect2: 
error: ld returned 1 exit status\n"

Do you know ho to solve this problem?

TacoGrandeTX, 2019-03-25:
@wlamprecht Can you provide an example of the logging calls that you are using?

wlamprecht, 2019-03-25:
@TacoGrandeTX 
My goal is to get the output from NRF_LOG_DEBUG, NRF_LOG_WARNING and NRF_LOG_ERROR 
e.g. in file :
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/TARGET_SOFTDEVICE_COMMON/softdevice/common/nrf_sdh_ble.c

NRF_LOG_WARNING("RAM starts at 0x%x, can be adjusted to 0x%x.",app_ram_start_link, *p_app_ram_start);
or
NRF_LOG_DEBUG("RAM starts at 0x%x", app_ram_start_link);

closed 2019-05-31 by ciarmcom