ARMmbed/mbed-os #9648: Read command respond with ATCmdParser

masoudr, 2019-02-09:
[* ] Question
    [*] Enhancement
    [ ] Bug
Hi,
I'm trying to use `ATCmdParser` driver to communicate with a Quectel GSM modem. I wrote a function to receive the whole response from the modem. But as there is only a single `recv` to read data with `scanf` format which lacks whitespace and newlines, it fails with a long response.
I tried to read input char by char which worked fine but I think it is not the best way to read a multiline response because there is no timeout on it.
Here is a minimal example of what I did.

```

#include "mbed.h"
#include "string"
#include "platform\ATCmdParser.h"
#include "drivers\UARTSerial.h"

UARTSerial modem_serial;
ATCmdParser parser;

bool hasEnding (std::string const &fullString, std::string const &ending) {
    if (fullString.length() >= ending.length()) {
        return (0 == fullString.compare (fullString.length() - ending.length(), ending.length(), ending));
    } else {
        return false;
    }
}

std::string Modem::read_until_string_basic(const char *value)
{
    std::string respond;
    int ch;
    while (((ch = parser.getc()) != EOF) && (!hasEnding(respond, value)))
    {
        respond += ch;
    }
    return respond;
}

std::string read_until_string(const char *value, int timeout)
{
    std::string respond;
    char buffer[1024];
    parser.set_timeout(timeout);
    while(parser.recv("%[^\n]%*c", buffer)){
        respond.append(buffer);
        if (hasEnding(respond, value))
            break;
        respond.append("\n");
    }
    parser.set_timeout(default_timeout);
    return respond;
}

int main()
{
    modem_serial(Tx, Rx, 115200);
    parser(&modem_serial)
    parser.debug_on(1);
    parser.set_delimiter("\r\n");
    parser.set_timeout(default_timeout);
    parser.send("AT+Something");
    printf("%s\r\n", read_until_string("OK", 1000).c_str())
}

```

kjbracey-arm, 2019-02-11:
`ATCmdParser` is designed to responses line-at-a-time. If you're processing a multi-line response, the normal structure would be 1 or more intermediate lines, and a final response line. This can be handled by doing `recv` for the expected final good line ("OK" or whatever), and making sure you have installed OOB handlers for the possible intermediates or alternate error responses.

Those OOB handlers can have a bit of code like "if processing AT+Something, set this flag", or "if processing AT+Something, transfer data to here", or "set error flag, abort current recv".

For an example, look at `ESP8266::_recv_tcp_passive`:

        _sock_i[id].tcp_data = (char*)data;
        _sock_i[id].tcp_data_rcvd = NSAPI_ERROR_WOULD_BLOCK;
        bool done = _parser.send("AT+CIPRECVDATA=%d,%lu", id, amount)
                                && _parser.recv("OK\n");
        _sock_i[id].tcp_data = NULL;
        if (_sock_i[id].tcp_data_rcvd > 0) {

The actual work is done by `ESP8266::_oob_tcp_data_hdlr` which catches the `+CPIPRECVDATA` line before the `OK`, copies data to `tcp_data` and updates `tcp_data_rcvd`.

masoudr, 2019-02-12:
@kjbracey-arm Thanks, but I need some way to read a whole response. For example, when I'm sending an HTTP GET request, I need to get the whole response that is completely unpredictable and can also have special characters or newlines. I was disappointed that this feature is somehow missing in the library and I think reading underlying stream char by char is totally useless and it would be better to just stick to directly reading with `UARTSerial`.

kjbracey-arm, 2019-02-12:
`ATCmdParser::read` will let you read raw data, assuming you know how many bytes to expect. (And that just passes through to the underlying `FileHandle::read`). The normal set-up would be that you use `recv` to find whatever lead-in prefix, which would usually contain a length indication, then use `read` to get the raw data.

masoudr, 2019-02-12:
@kjbracey-arm Do you mind edit `read_until_string` function with this?

kjbracey-arm, 2019-02-12:
Looking at your example, I'm not quite sure what the use case is. You are reading arbitrary data up until you see an "OK"? But what if the data contains "OK"?

There must be some sort of framing or escaping mechanism if arbitrary data is being transferred. Normally these sort of devices tell you the length of upcoming before giving you the data, so it's

     parser.recv("+CIPDATA:%d,", &len);
     parser.read(buffer, len);

And in the other direction it would be

    parser.send("ATSEND:%d,%*s\n", len, len, buffer);

closed 2019-02-26 by masoudr