ARMmbed/mbed-os #9665: LPC1768: running code with a JTAG probe stalls the code

WillyKaze, 2019-02-11:
### Description

I'm currently trying to debug my application with a JTAG probe. Everything works as excepted, I am able to access the registers, memory, variables and peripherals.
However the running code stop invariably in __semihost api after hitting a break-point in assembly.

```c
static inline int __semihost(int reason, const void *arg) {
    int value;

    asm volatile (
       "mov r0, %1"          "\n\t"
       "mov r1, %2"          "\n\t"
       AngelSWIInsn " %a3"   "\n\t"    // Stop here because AngelSWIInsn is defined to "bkpt"
       "mov %0, r0"
       : "=r" (value)                                         /* output operands             */
       : "r" (reason), "r" (arg), "i" (AngelSWI)              /* input operands              */
       : "r0", "r1", "r2", "r3", "ip", "lr", "memory", "cc"   /* list of clobbered registers */
    );

    return value;
}
```

I am debugging with MCU Eclipse after exporting the code from mbed-cli using
```
mbed export -m LPC1768 -i gnuarmeclipse --profile mbed-os/tools/profiles/debug.json
```

This behavior is observed with mbed-os **5.8.6** and **5.11.2**.
The code starts and runs fine without a debugger attached.

The JTAG probe used is a Digilent JTAG HS3, supported by openocd v0.10.0 provided by gnu-mcu-eclipse

Attached a screenshot of the debugging session:  
[image]

Unfortunately, I don't have any other JTAG probe to test this issue using a different on.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-02-11:
@WillyKaze This sounds like normal halt-before-run debug behavior. Is there something that makes you think that this should not be the case?

WillyKaze, 2019-02-12:
One thing that I didn't say is that I'm trying to run the code without uploading the firmware with openocd, it is already uploaded with a different command line utility.

0xc0170, 2019-02-12:
lpc1768 target supports semihosting (defined in targets.json file). Be aware, idle loop invokes sleep. This function disables interface prior the sleep

```
void hal_sleep(void) {

#if (DEVICE_SEMIHOST == 1)
    // ensure debug is disconnected
    mbed_interface_disconnect();
#endif
```
 from hal_sleep.c file. Is this affecting you?

kjbracey-arm, 2019-02-12:
Is semihosting disabled in OpenOCD? If semihosting is enabled, it should recognise and respond to that semihosting `BKPT` instruction without treating it as a visible breakpoint, but it seems like it's just treating it as a stop request.

But the correct response to that semihosting request would be, I believe, to disconnect the debugger. LP1768's hal_sleep call is for some reason deliberately making this debugger disconnect call. Or is that just disconnecting the semihosting function, not the intrusive debug?

So, my first two suggestions are:

* Ensure semihosting is on in OpenOCD
* Remove SEMIHOST from the LPC1768 section of targets.json

That should work if the sleep problem is only the semihosting. 

If it turns out that any sleep throws off the JTAG debugger, then you may need to mess with `hal_sleep` further. I would have hoped there would be some form of power setup for LPC1768 where `WFI` doesn't lose debug connection, but if it insists, just make it an empty function.

WillyKaze, 2019-02-12:
Removing SEMIHOST from the LPC1768 in targets.json worked for me. I also had to remove LOCALFILESYSTEM in order to get mbed to compile.

Semihosting option had no effect on the debugging session (before removing the SEMIHOST and after)

I will try to keep SEMIHOST enabled and make sure hal_sleep doesn't call mbed_interface_disconnect();

WillyKaze, 2019-02-12:
Commenting out mbed_interface_disconnect() in hal_sleep() does fix the issue.

I have trouble understanding why this might be required when entering sleep mode then

kjbracey-arm, 2019-02-12:
I don't know why. It's normal for devices like this to lose JTAG debug connection totally when entering deep sleep, but that "just happens". And deep sleep is suppressed because of that for debug profile builds, so your debug build will be using hal_sleep.

Conceivably if the device has semihost support, then it might want to prewarn the debugger for a clean semihost shutdown - maybe that's what the "disconnect" is. But then I would have thought it's only needed in deep sleep.

As your semihosting isn't active otherwise - it would only be used if you redirected the console up the JTAG or tried accessing files across it with `LocalFileSystem` - presumably the disconnect would achieve nothing and all it does is expose the fact that your semihosting calls aren't working.

@ARMmbed/team-nxp 

Could also be a flaw in OpenOCD - not sure how much variance if any there is in debugger support. Maybe that "disconnect" call isn't implemented in its semihosting support, and unimplemented calls just stop it?

cmonr, 2019-02-12:
@flit Would you happen to have any thoughts here?

kjbracey-arm, 2019-02-12:
This could be related to #6578 and #6624 - both open issues with faults happening in the LPC1768 semihost code, but without a debugger attached.

I think maybe you've solved those. As there is a "disconnect" in the sleep calls, that kills the `LocalFileSystem`, so first time the code ever idles, `LocalFileSystem` is dead.

WillyKaze, 2019-02-14:
That's actually mentioned in the comments above hal_sleep():

https://github.com/ARMmbed/mbed-os/blob/f3424da060734fa9e27bf6a798747fb48ad63938/targets/TARGET_NXP/TARGET_LPC176X/sleep.c#L20-L32

kjbracey-arm, 2019-02-14:
`LocalFileSystem` is specifically a thing for accessing a host filing system at the other end of the JTAG. Seems odd that it's called `HostFileSystem`, but I guess on the original Mbed LPC1768 board you were actually talking to an interface on the board which manipulated the SD card, not a PC debugger. ( https://os.mbed.com/handbook/LocalFileSystem )

So if `LocalFileSystem` is in use, you must have something attached via the JTAG. In this case it's the mbed interface, not a debugger - not sure how the two interact.

If the current `hal_sleep` for LPC1768 inherently kills `LocalFileSystem`, it seems there should be some mechanism to suppress even that sleep, just as we can suppress deep sleep.  Could be done automatically via `LocalFileSystem`'s constructor, which would cover most use cases.

Are you using one of these Mbed LPC1768 boards where `LocalFileSystem` works?

WillyKaze, 2019-02-15:
> Are you using one of these Mbed LPC1768 boards where `LocalFileSystem` works?

I have two of them but I don't use them to develop because there is a lot of IOs on our custom boards

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@WillyKaze This has become stale. Please reopen it if you still have questions.