ARMmbed/mbed-os #9672: Bring CI specific modules into requirements.txt?

cmonr, 2019-02-12:
### Description

@ARMmbed/mbed-os-tools @ARMmbed/mbed-os-maintainers 

Thoughts about bringing in the CI-only requirements permenantly into Mbed OS's requirements.txt?
Basically, moves https://github.com/ARMmbed/mbed-os/blob/master/.travis.yml#L111 into https://github.com/ARMmbed/mbed-os/blob/master/requirements.txt

CC @JuhPuur 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-02-12:
I initially thought this was the case until travis broke and we found out about python modules there.

Aren't there "test reqs" ? `install_requires` - requirements.txt, `test_requires` - test_requirements.txt , would this be good ? Rather than all in one?

bridadan, 2019-02-12:
Yeah what @0xc0170 said, please use a separate `test_requirements.txt`, this is a common practice (we use it in [mbed-os-tools](https://github.com/ARMmbed/mbed-os-tools/blob/master/test_requirements.txt)!).

cmonr, 2019-02-12:
😞 I don't like adding _even_ more things into the root of the project.

Will look into making a PR.

Although I'm wondering...

I don't think `pip install -r A.txt; pip install -r B.txt` will yield the same behavior as `<B.txt cat - >> A.txt; pip install -r A.txt`

bridadan, 2019-02-12:
> I don't like adding _even more_ things into the root of the project.

You can put it wherever you want, doesn't have to be in the root 😄 

> I suspect that running seperate `pip install -r` invocations could yield different dependencies than a single `pip install -r <merged_file>`.

Not 100% sure, but I'd think it'd be fairly reasonable. You can also supply multiple `-r` arguments to `pip install`:

```
$ pip install --help
...
Install Options:
  -r, --requirement <file>    Install from the given requirements file. This
                              option can be used multiple times.
...
```

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@cmonr It sounds like both @bridadan and @0xc0170 prefer separate file. I'll keep this closed unless you still feel strongly about your proposal, in which case, feel free to reopen it.