ARMmbed/mbed-os #9674: Unable to communicate with BLE APIs through greentea testing

UrmilaKL, 2019-02-12:
Description: Unable to communicate with BLE APIs through greentea testing
Target: DISCO_L475VG_IOT01A
Toolchain: GCC_ARM
mbed-cli version: 1.8.3
mbed-os->5.11.1
Steps to Reproduce:
Create directory structure like below
inside mbed-os -> TESTS->Group1->TestCase1->main.cpp(I have attached my main.cpp which i tried)
Create the BLE object like below
BLE &ble = BLE::Instance();
ble.gap().startAdvertising();
Add all necessary (BLE.h, BLE.cpp... ) files to the same directory where you are running the test case
Try to run the testcase using below command
mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM -n tests-Group1-TestCase1
Observation: Other existing test cases are passing, When we try to create BLE object and try to call any API it is failing with the below error
"Please provide the implementation for MBED BLE"
Please find the attached logs and main.cpp file
GreenTeaErrorForBLEAPI.txt
main.txt

UrmilaKL, 2019-02-12:
Hi,

Is there any test cases/API  related to BLE examples using testing green tea and icetea?

cmonr, 2019-02-13:
@ARMmbed/mbed-os-pan Could y'all help out here?

paul-szczepanek-arm, 2019-02-14:
what' sin your mbed_app.json?
You need the ble instance to be provided by someone
'''
        "DISCO_L475VG_IOT01A": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_BLUENRG"]
        },
'''
you can get the bluenrg here:
https://github.com/ARMmbed/cordio-ble-x-nucleo-idb0xa1/#811f3fea7aa8083c0bbf378e1b51a8b131d7efcc

jeromecoutant, 2019-02-20:
@ARMmbed/team-st-ble

UrmilaKL, 2019-03-05:
HI @paul-szczepanek,
I am using mbed os folder alone bu cloning it, So there is no mbed_app.json file in it for testing using GreenTea

Thanks
Urmila Bhat

paul-szczepanek-arm, 2019-03-05:
OK, then add one and put this inside (you will also need the repo from my link in the previous comment):
```
{
    "target_overrides": {
        "DISCO_L475VG_IOT01A": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_BLUENRG"]
        }
}
```
If it doesn't get pulled in automatically, use --app-config switch on the command line. You can put it in the test folder to limit it to the test you want to run.

pan-, 2019-03-05:
> Is there any test cases/API related to BLE examples using testing green tea and icetea?

Our internal test suite use Icetea to test BLE. 

Looking at your main file, I can tell that you need to rework your test architecture as BLE requires processing of its event. You need to provide a scheduler bound to the event queue used to process BLE  to `Harness`.

UrmilaKL, 2019-03-07:
Could you please provide us any small testcase for BLE using greentea/icetea ?
So that I can take that example and explore more on it

UrmilaKL, 2019-03-07:
Or could you please provide me one single BLE testcase where i can run it through greentea

LMESTM, 2019-03-08:
cc @avilei

ConradBraam, 2019-03-12:
I'm just observing this thread here @UrmilaKL , my function is maintaining test cases for this area. However unable to help much because the full test application is non-trivial, so not suitable for learning the frameworks, because it assumes you are very familiar with icetea already. What cases or scenarios specifically are you wanting to validate (test)?

UrmilaKL, 2019-03-12:
@ConradBraam,
I require any kind of BLE test case which should be able to communicate with any BLE API for any app is fine(EX: Heartrate, thermometer etc)

jeromecoutant, 2019-10-11:
Hi
Where is the current action?
- ARM greentea/test team ?
- ARM BLE team ?

@adbridge 
please remove the device st label :-)
Thx

adbridge, 2019-10-11:
@donatieng fyi

adbridge, 2020-10-15:
As nothing has happened on this for over 18 months, I'm going to close this. @UrmilaKL if this is still an issue for you please raise a new issue.

closed 2020-10-15 by adbridge