ARMmbed/mbed-os #9675: Difference between `-v` and `-vv`?

cmonr, 2019-02-12:
### Description

I'm hoping this is a really silly question on my part.
In taking a quick re-look at this (https://github.com/ARMmbed/mbed-os/issues/8146#issue-360754864) issue, I found that appending `-vv` instead of `-v` causes the linked issue's build to complete.

This is the only difference in the build tools:
``` diff
diff --git a/tools/build_api.py b/tools/build_api.py
index a6bb06add0..bb1575226b 100644
--- a/tools/build_api.py
+++ b/tools/build_api.py
@@ -390,7 +390,10 @@ def _fill_header(region_list, current_region):
                 ih = intelhex_offset(region_dict[data].filename, offset=region_dict[data].start)
             if subtype.startswith("CRCITT32"):
                 fmt = {"CRCITT32be": ">l", "CRCITT32le": "<l"}[subtype]
+                print(fmt)
+                print(zlib.crc32(ih.tobinarray()))
                 header.puts(start, struct.pack(fmt, zlib.crc32(ih.tobinarray())))
+                print("OK")
             elif subtype.startswith("SHA"):
                 if subtype == "SHA256":
                     hash = hashlib.sha256()
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-02-12:
@theotherjimmy @bridadan

theotherjimmy, 2019-02-12:
Well, it's parsed as `options.very_verbose` but I'm not sure of anywhere that is used.

bridadan, 2019-02-12:
I believe its just used in `mbed-cli`. Typically to show extra info when doing SCM commands, showing tracebacks in certain places, or for passing the normal verbose flag (`-v`) to the scripts in mbed-os.

cmonr, 2019-02-12:
Whoops, my mistake. Didn't successfully attached the files last night.

The reason I even ask is because the crc of the header is different depending on which flag is used, which I didn't think should be the case.

bridadan, 2019-02-12:
If you're talking about the header for the bootloader, I'm pretty sure it uses a timestamp: https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BOOTLOADER/mbed_lib.json#L8

cmonr, 2019-02-12:
Aaahhh, thank you! That would be the missing piece of knowledge I was missing.

Should be able to piece together any other oddities from there.

closed 2019-02-12 by cmonr