ARMmbed/mbed-os #9683: License documentation for non-apache files (non-targets code)

0xc0170, 2019-02-12:
### Description

https://github.com/ARMmbed/mbed-os/pull/9502 is adding license overview to our main Readme. We should add proper license documentation for each of item in the list there. This issue is covering non targets folders. Targets will be separated issue, external teams involved.

Our license page (section Using a different license): https://os.mbed.com/docs/mbed-os/latest/contributing/license.html

We need license files to describe the licenses in the componenent/module/driver.

This is the list with teams who can help to document the licenses.

- cmsis - MIT, BSD-3-Clause
- rtos - MIT
- drivers) - MIT

   @ARMmbed/mbed-os-core 

- features/cryptocell/FEATURE_CRYPTOCELL310- ARM Object Code and Header Files License

   @RonEld 

- features/FEATURE_BOOTLOADER - PBL
- features/frameworks/unity/unity - MIT
- tools/data/rpc- MIT

   @ARMmbed/mbed-os-tools / @ARMmbed/mbed-os-maintainers 

- features/lorawan - Revised BSD

    @ARMmbed/mbed-os-wan 

- features/lwipstack - BSD-style, MIT-style
- features/nanostack/sal-stack-nanostack - BSD-3-Clause
- features/netsocket/emac-drivers- BSD-style
- components/802.15.4_RF/mcr20a-rf-driver - BSD-3-Clause

   @ARMmbed/mbed-os-ipcore 

- features/storage - BSD-style, MIT

   @ARMmbed/mbed-os-storage 

- features/unsupported - MIT-style, BSD-style

- TESTS/mbed_hal/trng/pithy - BSD-3-Clause

   @ARMmbed/mbed-os-crypto 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

yanesca, 2019-02-12:
I think regarding `TESTS/mbed_hal/trng/pithy` @theamirocohen can give more relevant information than @ARMmbed/mbed-os-crypto .

0xc0170, 2019-02-12:
Thanks @yanesca , fixed

0xc0170, 2019-04-12:
This still needs to done. Very few folders contain license information.

bridadan, 2019-04-12:
@0xc0170 the @ARMmbed/mbed-os-tools team has removed the `tools/data/rpc` folder. Could you add some checkboxes to the issue to indicate the progress being made/what there is left to do?

0xc0170, 2019-04-15:
Updated, checkboxes created

0xc0170, 2019-08-14:
This can be closed. The license.md contains non target folders.

closed 2019-08-14 by 0xc0170