ARMmbed/mbed-os #9689: Redundant/conflicting arguments in test.py

bridadan, 2019-02-12:
### Description

The `test.py` script contains two arguments that are somewhat in conflict with each other: `--source` and `--paths`. The expected behavior is that all tests supplied within (potentially multiple) `--source` arguments would be picked up, but this is not the case. Instead you must supply them as `--paths`.

I propose to remove/deprecate the `--paths` argument in favor of the `--source` argument. This brings the `test.py` command line more inline with our other scripts. The `--paths` argument is not used by Mbed CLI so it should not affect most users.

Thoughts @Alex-EEE @sathysreenath @theotherjimmy ?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

Alex-EEE, 2019-02-12:
Yep, makes sense to me, especially since you can have multiple --source arguments, I don't really see what value --paths adds

Alex-EEE, 2020-10-13:
Hate to hear you left mbed @bridadan, we will miss you!

What target(s) are you using?  All targets
What toolchain(s) are you using?  All toolchains
What version of Mbed OS are you using (tag or sha)?  n/a AFAIK all versions of the toolchain still have both --paths and --source as arguments
It would help if you could also specify the versions of any tools you are using?
How can we reproduce your issue?  Invoke the CLI and try to figure out whether you want to use --paths or --source

adbridge, 2020-10-15:
This looks more like an enhancement request rather than an actual bug. As our tools have changed for Mbed 6 and there will be no updates for the Mbed 5 version of tools, I'm going to close this as a 'bug' and pass the details to the new tools team to look at checking against the new tools. @Patater can you consider this for the 6.x tools please (assuming it is an issue there).

closed 2020-10-15 by adbridge