ARMmbed/mbed-os #9695: NRF52_DK LED2 always on

byq77, 2019-02-12:
### Description

I have following problem with Nordic's nRF52 DK board and the newest mbed-os. Regardless of used code **the second on-board led is always on**. There was no such behavior in applications build with previous versions of mbed-os that used Nordic SDK 11 for this target. I checked my old mbed apps (mbed-os 5.8) and examples from Nordic SDK 14.2 and they don't have this issue so it's not hardware. 

The bug is not critical but annoying.

----

**Target:**
NRF52_DK

**Tools:**
mbed-cli 1.8.3

**Toolchain:**
GCC_ARM

**Toolchain version:**
GNU Tools for ARM Embedded Processors 6-2017-q2-update 6.3.1 20170620 (release) 

**Mbed OS SHA:**
a8d1d26fa7 (HEAD, tag: mbed-os-5.11.3, tag: latest)

**Steps to reproduce**
Build and run the following code:

```cpp
#include <mbed.h>
#include <Thread.h>
#include <gpio_api.h>
#include "ble/BLE.h"

DigitalOut led(LED1);
gpio_t led2;
int main()
{
    BLE &ble = BLE::Instance();
    ble.init();
    gpio_init(&led2,LED2);
    gpio_dir(&led2,PIN_OUTPUT);
    printf("Is connected: %d\r\n",gpio_is_connected(&led2));
    int led2_state = 0;
    while(1)
    {
        led = !led;
        gpio_write(&led2, led2_state^=1);
        printf("LED1: %d, LED2: %d\r\n",led.read(),gpio_read(&led2));
        ThisThread::sleep_for(1000);
    }
}
```

I used following .mbedignore file inside mbed-os directory:

```
features/cellular/*
features/cryptocell/*
features/deprecated_warnings/*
features/device_key/*
features/FEATURE_BOOTLOADER/*
features/frameworks/*
features/lorawan/*
features/lwipstack/*
features/nanostack/*
features/netsocket/*
features/nfc/*
features/unsupported/*
features/storage/*
components/wifi/*
components/802.15.4_RF/*
components/storage/*
components/TARGET_PSA/*
```

Logs look like this:

```
Is connected: 1
LED1: 1, LED2: 1
LED1: 0, LED2: 0
LED1: 1, LED2: 1
...
```
However LED2 is always on...

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

pan-, 2019-02-14:
Could you test without ble and provide your `mbed_app.json` ? 

@scartmell-arm @donatieng Debug code for Cordio LL has been removed, right ?

byq77, 2019-02-15:
My `mbed_app.json`:
```json
{
    "config": {},
    "macros": [],
    "target_overrides": {
        "NRF52_DK": {
            "events.shared-dispatch-from-application": 0,
            "platform.default-serial-baud-rate": 115200,
            "platform.stdio-baud-rate": 115200,
            "target.extra_labels_remove": [
                "SOFTDEVICE_X_FULL"
            ],
            "target.extra_labels_add": [
                "SOFTDEVICE_X_OTA"
            ]
        }
    }
}
```

Commenting lines
```cpp
    // BLE &ble = BLE::Instance();
    // ble.init();
```
has no impact. 

I use following task in Visual Studio Code to build the firmware:
```json
      {
            "label": "BUILD (RELEASE)",
            "type": "shell",
            "command": "mbed",
            "args": [
                "compile",
                "-m",
                "NRF52_DK",
                "-t",
                "GCC_ARM",
                "--profile",
                "release",
                "-N",
                "firmware",
                "--source",
                "${workspaceFolder}/../mbed-os",
                "--source",
                "${workspaceFolder}",
                "--build",
                "${workspaceFolder}/BUILD/RELEASE"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            },
            "problemMatcher": {
                "owner": "cpp",
                "fileLocation":"absolute",
                "pattern": {
                    "regexp": "^(\\[ERROR\\])*\\s*(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                    "file": 2,
                    "line": 3,
                    "column": 4,
                    "severity": 5,
                    "message": 6
                }
            }
        }
```

TacoGrandeTX, 2019-02-19:
@byq77 We've noticed this for a while now so thanks for escalating this.  The issue is caused by enabling the ITM for SWO usage as pin 18 (the SWO pin) happens to be mapped to LED2 on the nRF52_DK.  

We have made a PR to address this here:  https://github.com/ARMmbed/mbed-os/pull/9767

In the meantime, you can fix this in your mbed_app.json like this:

```
{
   "target_overrides": {
        "NRF52_DK": {
            "target.device_has_remove": ["ITM"]
        }
    }
}
```

byq77, 2019-02-20:
Many thanks !

TacoGrandeTX, 2019-02-22:
@byq77 The PR was merged and it will be available in v5.11.5 slated for release next week.

closed 2019-02-22 by TacoGrandeTX