ARMmbed/mbed-os #9699: Some build issues with PSA_Target on M2351

ccli8, 2019-02-13:
### Environment

[NuMaker-mbed-TZ-secure-example](https://github.com/OpenNuvoton/NuMaker-mbed-TZ-secure-example)
b996816 Update to mbed-os 5.10.2

mbed-os
709e6ff795 Merge pull request #9600 from kjbracey-arm/atomic_exchange_64

### Description

I am adding `PSA_Target` support into **NUMAKER_PFM_M2351** target and meet some issues with build tool. To re-produce the build issues, I make the following draft modifications. The compile process will fail, but this modification is just to show the build issues.

1. Add `SPE_Target` into inheritance list of NUMAKER_PFM_M2351 in **mbed-os/targets/target.json**:
    ```json
    "NUMAKER_PFM_M2351": {
        "core": "Cortex-M23-NS",
        "default_toolchain": "ARMC6",
        ......
        "overrides": {
            "mpu-rom-end": "0x1fffffff"
        },
        "inherits": ["SPE_Target", "Target"],
        ......
    ``` 

1. Override `target.secure-rom-start` and like symbols in **NuMaker-mbed-TZ-secure-example/mbed_app.json**:

    ```json
    "target_overrides": {
            "*": {
                "platform.stdio-baud-rate": 9600,
                "platform.stdio-convert-newlines": true
            },
            "NUMAKER_PFM_M2351": {
                "target.core": "Cortex-M23",
                "target.inherits": ["NUMAKER_PFM_M2351"],
                "target.device_has_remove": ["TRNG", "SERIAL", "SERIAL_ASYNCH", "SERIAL_FC", "STDIO_MESSAGES"],
            
                "target.secure-rom-start":    "0x0",
                "target.secure-rom-size":     "0x40000",
                "target.secure-ram-start":    "0x20000000",
                "target.secure-ram-size":     "0x8000"
            }
    }
    ```

Compile by running under **NuMaker-mbed-TZ-secure-example**:
```mbed compile -m NUMAKER_PFM_M2351 -t ARMC6```

And meet some issues:
1. `MBED_ROM_START` and like symbols are duplicated (see **BUILD/NUMAKER_PFM_M2351/ARMC6/.profile-ld**).
    ```json
    {
        "flags": [
            "--cpu=cortex-m23", 
            "--keep=os_cb_sections", 
            "--legacyalign", 
            "--predefine=\"-DDOMAIN_NS=0x1\"", 
            "--predefine=\"-DMBED_BOOT_STACK_SIZE=1024\"", 
            "--predefine=\"-DMBED_RAM_SIZE=0x18000\"", 
            "--predefine=\"-DMBED_RAM_SIZE=None\"", 
            "--predefine=\"-DMBED_RAM_START=0x20000000\"", 
            "--predefine=\"-DMBED_RAM_START=None\"", 
            "--predefine=\"-DMBED_ROM_SIZE=0x80000\"", 
            "--predefine=\"-DMBED_ROM_SIZE=None\"", 
            "--predefine=\"-DMBED_ROM_START=0x0\"", 
            "--predefine=\"-DMBED_ROM_START=None\"", 
            "--show_full_path"
        ],
        ......
    }
    ```
1. `mbed_app.json` is not consumed implicitly. `PSA_SECURE_ROM_START` and like symbols are not listed in **BUILD/NUMAKER_PFM_M2351/ARMC6/mbed_config.h**. This error disappears if I specify `--app-config mbed_app.json` explicitly instead:
    ```mbed compile -m NUMAKER_PFM_M2351 -t ARMC6 --app-config mbed_app.json```
1. `.mbedignore` doesn't work. For example, LoRa related files are listed in **.mbedignore**, but they are still added into compile list.
    ```
    Compile [ 16.1%]: LoRaWANInterface.cpp
    Compile [ 16.2%]: LoRaWANStack.cpp
    Compile [ 16.4%]: LoRaMacChannelPlan.cpp
    Compile [ 16.5%]: LoRaMac.cpp
    Compile [ 16.6%]: LoRaMacCrypto.cpp
    Compile [ 16.8%]: LoRaMacCommand.cpp
    Compile [ 16.9%]: LoRaPHYAS923.cpp
    Compile [ 17.1%]: LoRaPHY.cpp
    ```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

mikisch81, 2019-02-13:
@ARMmbed/mbed-os-psa 

Why are you adding `SPE_Target` to a non-secure v8m target?
`NUMAKER_PFM_M2351` has a `Cortex-M23-NS` core type so it builds the non-secure v8m target.

But the current PSA solution in `master` branch only tackles dual-core targets (Such as Future-Sequana) and not v8m.
#9653 (along with #9221) add support for PSA on v8m by importing TF-M (which has v8m PSA support).

Please see [Future-Sequana PSA targets](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L7988-L8031) for reference for the current dual-core PSA target configuration, but again it is relevant only to dual core.

ccli8, 2019-02-13:
Re-check **NuMaker-mbed-TZ-secure-example/mbed_app.json**,  CPU core is changed to secure target `Cortex-M23`. This is the old means to build M33/M23 secure image before PSA is introduced. And this means still works with the latest mbed-os. Actually, I am adding PSA/TFM support to M2351 based on #9221. These issues are what I met during this port. So I raised them separately. The changes mentioned above are just to easily show these build issues, and won't get into my formal port.

```json

"target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 9600,
            "platform.stdio-convert-newlines": true
        },
        "NUMAKER_PFM_M2351": {
            "target.core": "Cortex-M23",
            "target.inherits": ["NUMAKER_PFM_M2351"],
            "target.device_has_remove": ["TRNG", "SERIAL", "SERIAL_ASYNCH", "SERIAL_FC", "STDIO_MESSAGES"],

```

mikisch81, 2019-02-13:
The secure side build process is completely different than regular mbed-os build.
It is going to be built as bare-metal (PRs #9561 and #9628).

There will be no no Mbed-OS running in the secure side at all, only TF-M and specific drivers will be built, you can look at [TF-M integration guide](https://github.com/ARMmbed/mbed-os/blob/573acad133725ff847f4dc261b1d136159b09122/components/TARGET_PSA/TARGET_TFM/tf-m-integration.md) from #9653 with instructions on how to port a new target with TF-M to Mbed-OS (note that this guide is still a work-in-progress).

ccli8, 2019-02-14:
Checking #9221, it seems that I need to specify source paths through `--source` options to compile secure side code. When I specify `--source` options, `mbed_app.json` is not consumed implicitly and `mbedignore` doesn't work as issues 2/3 above. Is this behavior well-defined?

For the duplicate `MBED_ROM_START` symbol issue, I think it is common for targets inheriting `SPE_Target`. This symbol is generated in two places (both in **mbed-os/tools/toolchains/__init__.py**):
1. `MBED_ROM_START` comes from `target.secure-rom-start`
    ```python
    def add_linker_defines(self):
        ......
        flags2params = {}
        if self.target.is_PSA_non_secure_target:
            flags2params = {
                "MBED_ROM_START": "target.non-secure-rom-start",
                "MBED_ROM_SIZE": "target.non-secure-rom-size",
                "MBED_RAM_START": "target.non-secure-ram-start",
                "MBED_RAM_SIZE": "target.non-secure-ram-size"
            }
        if self.target.is_PSA_secure_target:
            flags2params = {
                "MBED_ROM_START": "target.secure-rom-start",
                "MBED_ROM_SIZE": "target.secure-rom-size",
                "MBED_RAM_START": "target.secure-ram-start",
                "MBED_RAM_SIZE": "target.secure-ram-size"
            }

        for flag, param in flags2params.items():
            define_string = self.make_ld_define(flag, params[param].value)
            self.ld.append(define_string)
            self.flags["ld"].append(define_string)
    ```
1. `MBED_ROM_START` comes from CMSIS memory spec
    ```python
    def _add_all_regions(self, region_list, active_region_name):
        for region in region_list:
            self._add_defines_from_region(region)
            if region.active:
                for define in [
                        ("%s_START" % active_region_name, "0x%x" % region.start),
                        ("%s_SIZE" % active_region_name, "0x%x" % region.size)
                ]:
                    define_string = self.make_ld_define(*define)
                    self.ld.append(define_string)
                    self.flags["ld"].append(define_string)
    ```
MUSCA_A1 neither has its CMSIS pack integrated into mbed-os nor defines `target.mbed_rom_start` symbol, so it doesn't meet the duplicate `MBED_ROM_START` issue as my M2351 example above.

alzix, 2019-05-12:
I think the issue was addressed on master.

Now we can configure and build both secure and non-secure images for V8-M target. 
Musca A1 support is added in #9221.
In addition there is LPC55S69 target that uses current build system.

@ccli8 - is this issue still relevant for you? If not, can we close it?

ccli8, 2019-05-13:
@alzix I re-check and find all the issues are fixed on master. Thanks for your information.

closed 2019-05-13 by ccli8