ARMmbed/mbed-os #9703: AT_CellularNetwork::read_reg_params_and_compare should copy lac on cellid change

NeilMacMullen, 2019-02-13:
Commit a477354292a3c
AT_CellularNetwork.cpp contains this code...

```
 if (reg_params._cell_id != -1 && reg_params._cell_id != _reg_params._cell_id) {
            _reg_params._cell_id = reg_params._cell_id;
           data.status_data = reg_params._cell_id;
	    _connection_status_cb((nsapi_event_t)CellularCellIDChanged, (intptr_t)&data);
        }
```
However, in the case where the modem has just started up it is possible for the initial population of _reg_params to set _cell_id and _lac to -1 (there is also a race condition between +CREG and +CGREG network status indications).  A subsequent network registration or cell-change will not set the _lac.

Adding 
`_reg_params._lac = reg_params._lac;`
to the block appears to fix the problem.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cmonr, 2019-02-13:
@ARMmbed/mbed-os-wan

closed 2019-04-11 by ciarmcom

0xc0170, 2019-04-11:
Fixed via ^^ PR