ARMmbed/mbed-os #9704: BUFF_SIZE in ATHandler.h is too low to avoid overflows

NeilMacMullen, 2019-02-13:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-02-13:
@ARMmbed/team-ublox

mburumaxwell, 2019-03-06:
This is also happening to Quectel BG96 when `+CGADDR: 1,"100.114.84.1"` is recevied. Is there any known remedy?

NeilMacMullen, 2019-03-06:
@mburumaxwell You can make a local patch to change BUFF_SIZE in AtHandler.h to 32 or a larger value.

bqam-ublox, 2019-03-07:
The local patch to change BUFF_SIZE can be used as a workaround. 
As far as I understand @cmonr , ARM shall provide a way to override a MACRO (linked from the json file similar to what is done in for example: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json)

closed 2019-04-11 by ciarmcom

0xc0170, 2019-04-11:
https://github.com/ARMmbed/mbed-os/pull/10362 should fixed this

mirelachirica, 2019-04-11:
> The Ublox G450 modem I am using sends this URC when setting a profile...
> 
> +UUPSDA: 0,"172.33.22.180"
> 
> This is larger than the 16 bytes defined to BUFF_SIZE in AtHandler.h and triggers an "AT overflow" warning.

What BUFF_SIZE should accomodate is just the URC's prefix length. Only the prefix needs to be checked for match. Rest of URC is read through AT read routines. In this case URC's prefix if "+UUPSDA:" so its length is 8 => 16 bytes is enough for this particular URC.

"AT overflow" is triggered when bytes from buffer are not read and fill_buffer is bringing in new data.

Root cause analysis for the "AT overflow" is needed. Why is the buffer content not read.
resp_stop() maybe missing.

horeich, 2022-06-30:
Thanks for pointing out the problem here, @mirelachirica 

This is a major flaw, the `_recv_buff` variable should be able to hold considerabily more characters than the `_cmd_buffer` buffer. For former we should define a buffer size of AT LEAST the size of the buffer which is defined in BufferedSerial (which is 256 by default) to be able to hold larger responses; in addition, as far as I understand the `BufferedSerial `class, we can read junks that are larger than the buffer defined in `BufferedSerial `as the buffer can be filled while reading. So we can end up with "AT overflow" although the buffer in the `ATHandler `class is as large as the one in the BufferedSerial class. 

For the command buffer 16 should be enough, as it only holds the AT command itself. I don't know why it was changed to 32 in the latest version. It won't solve the "AT overflow" problem!

Proposed solution: Define two buffer sizes, one for the receive AT buffer, one for the AT command buffer