ARMmbed/mbed-os #9711: UBLOX_AT_CellularContext appears to use wrong authentication for Stream sims

NeilMacMullen, 2019-02-13:
### Description

I don't know whether this is really a bug in the Stream network but attempting to connect using a Stream (ARM) sim with 

`iface->set_credentials("stream.co.uk", "streamip", "streamip");`

fails unless PAP is used in the UBLOX_AT_CellularContext constructor....

```
UBLOX_AT_CellularContext::UBLOX_AT_CellularContext(ATHandler &at, CellularDevice *device, const char *apn, bool cp_req, bool nonip_req) :
    AT_CellularContext(at, device, apn, cp_req, nonip_req)
{
    // The authentication to use
    //PAP seems to be required as the default in order for STREAM sims to work
	_auth = NSAPI_SECURITY_PAP;
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

elm3, 2019-03-25:
@ARMmbed/team-ublox

fahim-ublox, 2019-05-09:
What target you are using ? and if you have logs that will be helpful

NeilMacMullen, 2019-05-09:
@fahim-ublox It's a custom mbed target.  The modem is G450.  I'll see if I can collect logs next time I'm working on this but it may be a while.

jarvte, 2019-05-13:
@ciarmcom this can be linked to this issue: https://github.com/ARMmbed/mbed-os/issues/10551
We will add an API for changing the authentication type.

0xc0170, 2019-05-13:
> @ciarmcom this can be linked to this issue: #10551

I don't follow? Please link it in the jira

wajahat-ublox, 2019-05-20:
@0xc0170 
#10551 will not fully address this issue. Once an API is provided for application to be able to set authentication type, we will be using `_authentication_type` in our class to provide the said functionality.

mudassar-ublox, 2019-07-12:
#10877 updates auth configuration and you can use [set_authentication_type](https://github.com/ARMmbed/mbed-os/blob/d53bd6120dac490fceb225cc85ff781c33798731/features/cellular/framework/device/CellularContext.cpp#L85) to set `auth` type.

wajahat-ublox, 2019-07-12:
@NeilMacMullen Can you please have a look and see if #10877 fixes this issue?

loverdeg-ep, 2019-07-12:
Is their any particular reason that the SIM card doesn't provide all this information.
Embedded cellular seems to require a lot of manual configuration...

mudassar-ublox, 2019-09-03:
@NeilMacMullen any update regarding this issue, solved?

0xc0170, 2019-09-03:
I'll close this as resolved, can be reopened anytime.

closed 2019-09-03 by 0xc0170