ARMmbed/mbed-os #9712: Ublox G450 modem may need special-case code in UBLOX_AT_CellularContext::activate_profile

NeilMacMullen, 2019-02-13:
### Description
When connecting from a cold boot, UBLOX_AT_CellularContext::activate_profile issues some profile-configuration commands before finally issuing AT+UPSDA=0,3 to activate the profile.  On the G450 modem, this returns OK immediately but it appears that the profile is not actually active until a +UUPSDA URC has been received.  The following code appears to address this issue but I'm not sure whether it is G450-specific.
```

 if (_at.get_last_error() == NSAPI_ERROR_OK) {
                    // Activate, wait upto 30 seconds for the connection to be made
                    _at.set_at_timeout(30000);
                    _at.cmd_start("AT+UPSDA=0,3");
                    _at.cmd_stop();
                    _at.resp_start();
                    _at.resp_stop();					
					_at.resp_start("+UUPSDA:"); //wait for URC
					
                    _at.restore_at_timeout();

                    if (_at.get_last_error() == NSAPI_ERROR_OK) {
                        activated = true;
                    }
                }

```<!--
    Required
    Add detailed description of what you are reporting.
    Good example: https://os.mbed.com/docs/mbed-os/latest/contributing/workflow.html
    Things to consider sharing:
    - What target does this relate to?
    - What toolchain (name + version) are you using?
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
    - What is the SHA of Mbed OS (git log -n1 --oneline)?
    - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)
-->

### Issue request type

<!--
    Required
    Please add only one X to one of the following types. Do not fill multiple types (split the issue otherwise.)
    Please note this is not a GitHub task list, indenting the boxes or changing the format to add a '.' or '*' in front
    of them would change the meaning incorrectly. The only changes to be made are to add a description text under the
    description heading and to add a 'x' to the correct box.
-->
    [ ] Question
    [ ] Enhancement
    [x] Bug

elm3, 2019-03-25:
@ARMmbed/team-ublox , have you seen this?

fahim-ublox, 2019-03-26:
Internal Jira ticket : PGM_NO_MBD-216

bqam, 2019-03-26:
@NeilMacMullen thanks for the input. I have checked with SARA-G450 FW:09.02,A02.01 and I dont see any problem if I dont wait for the +UUPSDA URC. Infact, this URC is present for U201 as well and is received as soon as the UPSDA command is sent (for both modules). I checked with m-center (AT terminal) and I also used UPING command immediately after the context activation, able to see ping responses from 8.8.8.8. Please let me know how I can replicate this issue.

Moreover, as you might already know, the [cellular state machine](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/device/CellularStateMachine.cpp#L552) in mbed always makes sure that modem wakes up, initializes, perform SIM checks and registers on the network. For this reason, the URC can be ignored but it makes perfect sense to parse it for result (and in that case, we will be waiting for it as well).

bqam-ublox, 2019-03-26:
@NeilMacMullen thanks for the input. I have checked with SARA-G450 FW:09.02,A02.01 and I dont see any problem if I dont wait for the +UUPSDA URC. Infact, this URC is present for U201 as well and is received as soon as the UPSDA command is sent (for both modules). I checked with m-center (AT terminal) and I also used UPING command immediately after the context activation, able to see ping responses from 8.8.8.8. Please let me know how I can replicate this issue.

Moreover, as you might already know, the [cellular state machine](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/device/CellularStateMachine.cpp#L552) in mbed always makes sure that modem wakes up, initializes, perform SIM checks and registers on the network. For this reason, the URC can be ignored but it makes perfect sense to parse it for result (and in that case, we will be waiting for it as well).

NeilMacMullen, 2019-03-26:
@bqam-ublox  Thanks for looking at this.  IIRC, the error that occurs is that if you try to set up a socket without waiting for the UUPDSA URC, you get a CME ERROR (can't remember which one - probably invalid operation).   Since the logical flow is context-

bqam-ublox, 2019-03-27:
@NeilMacMullen I understand your point plus we need to be sure if the context is activated, will generate a pull request for review.

bqam-ublox, 2019-05-22:
[PR#10636](https://github.com/ARMmbed/mbed-os/pull/10636) generated. This should fix the issue. Please note that this PR is not tested on G450 but should be applicable on all modems.

MarceloSalazar, 2019-06-19:
Tested by Ublox and agreed to closed.

closed 2019-06-19 by MarceloSalazar