ARMmbed/mbed-os #9716: Inconsistent symbol generation with target.mbed_ram_start override

ccli8, 2019-02-14:
### Enviroonment

**mbed-os-example-blinky**
7cc5542 Merge pull request #161 from adbridge/master

**mbed-os**
f2abdcbcc6 Merge pull request #9639 from michalpasztamobica/esp8266_connect_timeout

### Description

I would like to override memory spec of cmsis pack on NUMAKER_PFM_M2351 but meet inconsistent symbol generation behavior when overriding `target.mbed_ram_start`/`target.mbed_ram_start`. To re-produce my test results, run `mbed compile -m NUMAKER_PFM_M2351 -t ARMC6` under **mbed-os-example-blinky** directory.

1. No override and then compile
    Get:
    **BUILD/NUMAKER_PFM_M2351/ARMC6/.profile-c**:
    ```python
    {
        "flags": [
            "-D__ASSERT_MSG", 
            "-std=gnu99", 
            "--target=arm-arm-none-eabi", 
            "-DMBED_RAM_SIZE=0x18000", 
            "-DMBED_RAM_START=0x20000000", 
            "-DMBED_ROM_SIZE=0x80000", 
            "-DMBED_ROM_START=0x0", 
    ```
    `MBED_RAM_START`/`MBED_RAM_SIZE` are generated from cmsis pack.

1. Override and then compile
    Override rom/ram spec:
    **mbed_app.json**:
    ```json
    {
        "target_overrides": {
            ......
            "NUMAKER_PFM_M2351": {
                "target.mbed_rom_start":    "0x10040000",
                "target.mbed_rom_size":     "0x40000",
                "target.mbed_ram_start":    "0x30008000",
                "target.mbed_ram_size":     "0x10000"
    ```
    Get:
    **BUILD/NUMAKER_PFM_M2351/ARMC6/.profile-c**:
    ```python
    {
        "flags": [
            "-D__ASSERT_MSG", 
            "-std=gnu99", 
            "--target=arm-arm-none-eabi", 
            "-DAPPLICATION_ADDR=0x10040000", 
            "-DAPPLICATION_RAM_ADDR=0x30008000", 
            "-DAPPLICATION_RAM_SIZE=0x10000", 
            "-DAPPLICATION_SIZE=0x40000", 
            "-DMBED_ROM_SIZE=0x40000", 
            "-DMBED_ROM_START=0x10040000", 
    ```
    `APPLICATION_RAM_ADDR`/`APPLICATION_RAM_SIZE` are generated from `target.mbed_ram_start`/`target.mbed_ram_size` overrides, but `MBED_RAM_START`/`MBED_RAM_SIZE` are not generated.

1. Override + change to `get_all_active_memories`
    In **mbed-os/tools/config/__init__.py**, if I comment out `if (self.has_ram_regions):` in `def get_all_active_memories`:
    **mbed-os\tools\config\__init__.py**:
    ```python
    def get_all_active_memories(self, memory_list):
        """Get information of all available rom/ram memories in the form of dictionary
        {Memory: [start_addr, size]}. Takes in the argument, a list of all available
        regions within the ram/rom memory"""
        # Override rom_start/rom_size
        #
        # This is usually done for a target which:
        # 1. Doesn't support CMSIS pack, or
        # 2. Supports TrustZone and user needs to change its flash partition

        ......

        for memory in valid_memories:
            mem_start, mem_size = self._get_mem_specs(
                [memory],
                cmsis_part,
                "Not enough information in CMSIS packs to build a bootloader "
                "project"
            )
            if memory=='IROM1' or memory=='PROGRAM_FLASH':
                mem_start = getattr(self.target, "mbed_rom_start", False) or mem_start
                mem_size = getattr(self.target, "mbed_rom_size", False) or mem_size
                memory = 'ROM'
            elif memory == 'IRAM1' or memory == 'SRAM_OC' or \
                memory == 'SRAM_UPPER' or memory == 'SRAM':
                # Comment out for generating MBED_RAM_SIZE/MBED_RAM_START symbols
                #if (self.has_ram_regions):
                #    continue
                mem_start = getattr(self.target, "mbed_ram_start", False) or mem_start
                mem_size = getattr(self.target, "mbed_ram_size", False) or mem_size
                memory = 'RAM'
    ```
    Get:
    **BUILD/NUMAKER_PFM_M2351/ARMC6/.profile-c**:
    ```json
    {
        "flags": [
            "-D__ASSERT_MSG", 
            "-std=gnu99", 
            "--target=arm-arm-none-eabi", 
            "-DAPPLICATION_ADDR=0x10040000", 
            "-DAPPLICATION_RAM_ADDR=0x30008000", 
            "-DAPPLICATION_RAM_SIZE=0x10000", 
            "-DAPPLICATION_SIZE=0x40000", 
            "-DMBED_RAM_SIZE=0x10000", 
            "-DMBED_RAM_START=0x30008000", 
            "-DMBED_ROM_SIZE=0x40000", 
            "-DMBED_ROM_START=0x10040000", 
    ```
    Both `APPLICATION_RAM_ADDR`/`APPLICATION_RAM_SIZE` and `MBED_RAM_START`/`MBED_RAM_SIZE` are generated from `target.mbed_ram_start`/`target.mbed_ram_size` overrides. But `MBED_RAM_START`/`MBED_RAM_SIZE` are duplicated in **BUILD/NUMAKER_PFM_M2351/ARMC6/.profile-ld**:
    ```json
    {
        "flags": [
            "--cpu=cortex-m23", 
            "--keep=os_cb_sections", 
            "--legacyalign", 
            "--predefine=\"-DDOMAIN_NS=0x1\"", 
            "--predefine=\"-DMBED_APP_SIZE=0x40000\"", 
            "--predefine=\"-DMBED_APP_START=0x10040000\"", 
            "--predefine=\"-DMBED_BOOT_STACK_SIZE=4096\"", 
            "--predefine=\"-DMBED_RAM_SIZE=0x10000\"", 
            "--predefine=\"-DMBED_RAM_SIZE=0x10000\"", 
            "--predefine=\"-DMBED_RAM_START=0x30008000\"", 
            "--predefine=\"-DMBED_RAM_START=0x30008000\"", 
            "--predefine=\"-DMBED_ROM_SIZE=0x40000\"", 
            "--predefine=\"-DMBED_ROM_START=0x10040000\"", 
    ```

What I expect is that `MBED_RAM_START`/`MBED_RAM_SIZE` are available in both **.profile-c** (for source files) and **.profile-ld** (for linker files) regardless of override.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-02-14:
@mprse Something you could review?

mprse, 2019-03-20:
> @mprse Something you could review?

I think the tools team should review this.

0xc0170, 2019-03-20:
@ARMmbed/mbed-os-tools

theotherjimmy, 2019-03-20:
I think #10113 and #10157 should have taken care of some of this.

0xc0170, 2019-03-21:
@ccli8 Please can you retest?

ccli8, 2019-03-21:
They don't fix the issue. Actually, #10008 is raised to fix the issue.

theotherjimmy, 2019-03-21:
Still a problem on the latest master.

PR #10008 (2019-03-27): Fix MBED_RAM_START/MBED_RAM_SIZE symbol generation
### Description

This PR tries to fix #9716 with build tool-generated `MBED_RAM_START`/`MBED_RAM_SIZE` symbols passed to C/C++ and linker files. It has the effects:
1. Fix `MBED_RAM_START`/`MBED_RAM_SIZE` symbols are not passed to C/C++ file (see build tool-generated **.profile-c**/**.profile-cxx**) when there are `target.mbed_ram_start`/`target.mbed_ram_size` overrides.
1. Avoid duplicate `MBED_RAM_START`/`MBED_RAM_SIZE` symbols passed to linker file (see build tool-generated **.profile-ld**).

### Pull request type

    [X] Fix
    [ ] Refac [... cut]
closed 2019-03-27 by cmonr