ARMmbed/mbed-os #9717: Allow defining bootloader target parameters as MACRO values

teetak01, 2019-02-14:
### Description

Currently if I use custom bootloader target I need to specify the addresses as hex-numbers.:
```
            "target.header_offset": "0x8000",
            "target.app_offset"   : "0x8400",
```

It would be helpful if I could instead use MACROs here, which would also improve the readability of the configuration.

```
    "macros": [
        "MBED_BOOTLOADER_SIZE=(32*1024)"
    ],
...
"MY_TARGET": {
            "target.header_offset": "(MBED_ROM_START)",
            "target.app_offset"   : "(MBED_ROM_START + MBED_BOOTLOADER_SIZE)"
```

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

cmonr, 2019-02-27:
CI started, although I'm not sure it's actually needed

theotherjimmy, 2019-03-22:
@teetak01 You proposed thing, placing the header _offset_ at the `MBED_ROM_START` _address_ will not do what you think it will. You want `0` instead.

Further, as C macros may be Turing complete (surprisingly) resolving the bootloader mode is as undecidable as the halting problem.

I don't have a problem with allowing a small amount of arithmetic to these defines, but allowing the use of C macros is not possible.

closed 2019-03-22 by theotherjimmy