ARMmbed/mbed-os #9720: STM32: sscanf causes hard fault exception when parsing hex bytes

seppestas, 2019-02-14:
### Description

Trying to parse a hex string using `sscanf` causes a hardfault exception on STM32 devices. I first encountered this with a custom board based on the STM32F091CB MCU (using the NUCLEO_F091RC mbed target) and I was able to reproduce this on the NUCLEO_F413ZH (STM32F413ZH) target/devboard.

The same script runs fine on my host computer and the EFM32LG_STK3600 (EFM32LG990F256) and NRF52_DK (NRF52832) targets/devboards.

#### Environment:
host: MacOS 10.14.2 (18C54)
compiler: arm-none-eabi-gcc version 6.3.1 20170620
mbed-os version: 5.9.5 (f8b140f8d7cb226e41486c5df66ac4f3ce699219) and 5.11.4 (ecb3c8c837)
targets: STM32, tested on NUCLEO_F413ZH and STM32F091CB based device with NUCLEO_F091RC target
profile: tested with both default and debug profiles

Example code:

```c++
#include "mbed.h"
#define dbg(fmt, ...) printf(fmt, ##__VA_ARGS__)

int main() {
    const char* response = "68656C6C6F2C20776F726C64";
    uint8_t payload[20] = {0};

    for (size_t i = 0; i < strlen(response)/2; i++) {
        dbg("%d:", i);
        if (sscanf(&response[i*2], "%02hhX", &payload[i]) != 1) {
            dbg("\r\nFailed to parse payload, last byte: %c%c\r\n", &response[i], &response[i+1]);
            return -1;
        }
        dbg("%c\r\n", payload[i]);t
    }

    while(true);
}
```

Backtrace (NUCLEO_F091RC):

```
(gdb) bt
#0  0x080014e6 in mbed_halt_system () at ./mbed-os/platform/mbed_error.c:68
#1  0x080016b0 in mbed_error (error_status=-2130771651, error_msg=0x8014920 "Fault exception",
    error_value=134269400, filename=0x0, line_number=0) at ./mbed-os/platform/mbed_error.c:209
#2  0x080024c6 in mbed_fault_handler (fault_type=16, mbed_fault_context_in=0x200018c8 <mbed_fault_context>,
    osRtxInfoIn=0x200001d0 <osRtxInfo>) at ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/mbed_rtx_fault_handler.c:69
#3  0x0800241c in Fault_Handler () at except.S:188
```

Backtrace (NUCLEO_F413ZH):

```
#0  0x08010f48 in __ssprint_r ()
#1  0x0800a75c in _svfprintf_r ()
#2  0x0800df58 in _vsnprintf_r ()
#3  0x0800dfb2 in vsnprintf ()
#4  0x08001468 in mbed_error_vfprintf (format=0x8012cd0 '\377' <repeats 200 times>..., arg=...)
    at ./mbed-os/platform/mbed_board.c:68
#5  0x08001444 in mbed_error_printf (format=0x8012cd0 '\377' <repeats 200 times>...)
    at ./mbed-os/platform/mbed_board.c:58
#6  0x080024dc in mbed_fault_handler (fault_type=16, mbed_fault_context_in=0x200015e0 <mbed_fault_context>,
    osRtxInfoIn=0x200002e8 <osRtxInfo>) at ./mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/mbed_rtx_fault_handler.c:38
#7  0x080024ba in Fault_Handler () at except.S:188
```

Weirdly enough the following code works fine (always storing value at index 0 in `payload`):

```c++
#include "mbed.h"
#define dbg(fmt, ...) printf(fmt, ##__VA_ARGS__)

int main() {
    const char* response = "68656C6C6F2C20776F726C64";
    uint8_t payload[20] = {0};

    for (size_t i = 0; i < strlen(response)/2; i++) {
        dbg("%d:", i);
        if (sscanf(&response[i*2], "%02hhX", &payload[0]) != 1) {
            dbg("\r\nFailed to parse payload, last byte: %c%c\r\n", &response[i], &response[i+1]);
            return -1;
        }
        dbg("%c\r\n", payload[0]);t
    }

    while(true);
}
```

And following code also doesn't work (reading in a character):

```c++
#include "mbed.h"
#define dbg(fmt, ...) printf(fmt, ##__VA_ARGS__)

int main() {
    const char* response = "68656C6C6F2C20776F726C64";
    char c;

    for (size_t i = 0; i < (strlen(response)-1)/2; i++) {
        dbg("%d:", i);
        if (sscanf(&response[i*2], "%02hhX", &c) != 1) {
            dbg("\r\nFailed to parse payload, last byte: %c%c\r\n", &response[i], &response[i+1]);
            return -1;
        }
        dbg("%c\r\n", c);
    }

    while(true);
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2019-02-14:
Some of the libraries do not support all C99 extensions, and `scanf("%hhX")` is one of them.  

I suspect it ends up treating it as "%02X", then tries doing an unaligned 32-bit write into your character array.  Cortex-M0 does not support unaligned writes, so it works if you only access the aligned first character.

To work around the library limitation, you can use `scanf("%02X", &an_int)` and then assign that integer to your char array.

In this particular case it might be better to use `strtoul` - avoids this issue and will likely pull in much less library code than `sscanf`.

seppestas, 2019-02-14:
Mhh, seems to be related to https://bugs.launchpad.net/gcc-arm-embedded/+bug/1399224.

@kjbracey-arm Thanks for suggesting the strtoul approach, I will look into it. I'm already using sscanf in other places in the code so the library code was already pulled in.

seppestas, 2019-02-14:
Hmm, the problem with `strtoul` is that it doesn't allow me to control the amount of characters parsed. It would also make dealing with the last bytes difficult if `response` is not aligned to an unsigned long. I came up with my own algorithm to parse the bytes character by character instead.

Anyway, I was able to confirm that the newlib I'm using was indeed compiled without the support for the required C99 functionality. I guess the other targets just didn't hardfault because of the unaligned writes.

Closing this issue since it is not caused by mbed-os afaik. It would however be useful to have a dedicated documentation page for these type of functions indicating the limitations in embedded systems.

closed 2019-02-14 by seppestas