ARMmbed/mbed-os #9723: ATHandler never clears sigio on filehandle

NeilMacMullen, 2019-02-14:
### Description
ATHandler.cpp uses sigio to drive the event loop for out-of-band AT traffic.  This is set in 

```
void ATHandler::set_filehandle_sigio()
{
    _fileHandle->sigio(Callback<void()>(this, &ATHandler::event));
}
```
However, there is no matching clear operation in the destructor.  This means that this pattern can run into problems...
```

UARTSerial * serial = new UARTSerial(NC, NC, 0);
UBLOX_AT * dev = new UBLOX_AT(serial);
CellularContext* iface = dev->create_context(serial, "stream.co.uk", false, false);
  ...
dev->delete_context(iface);
delete dev;
delete serial;
```
Because the modem can still be generating AT traffic (e.g. CREG indications) after the context has been deleted, causing the sigio callback to be called on a deleted object.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cmonr, 2019-02-15:
@ARMmbed/mbed-os-wan

AriParkkila, 2019-03-01:
This can be closed. Fix PR #9728 merged.

closed 2019-03-01 by 0xc0170