ARMmbed/mbed-os #9725: Memory leak in SecureStorage, when init() fails

RonEld, 2019-02-14:
### Description

There is a memory leak, in case `SecureStore.init()` fails:

`_entropy`, `_scratch_buf` and `_inc_set_handle` are allocated in the init function, and after that, `_underlying_kv->init()` and `_rbp_kv->init()` are called. If one of these underlying init functions fail, the flow goes to `fail`, skipping assignment of `_is_initialized` to true.

In the `deinit()` function, the heap allocated members that were allocated in the `init()` function` will be deleted only if `_is_initialized` initialized is true, which will result in a memory leak in case `init()` failed.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

aashishc1988, 2019-02-14:
@davidsaada

cmonr, 2019-02-15:
@ARMmbed/mbed-os-storage

michalpasztamobica, 2019-08-12:
@cmonr , seems that this issue is resolved by @davidsaada . Can we close it?

closed 2019-08-12 by 0xc0170