ARMmbed/mbed-os #9726: osDelay gets stuck on FUTURE_SEQUANA_M0_PSA in spm tests

orenc17, 2019-02-14:
### Description
In spm-smoke and spm-server there are calls to osDelay() which never return on the M0+ core
Removing MBED_TICKLESS does not seem to help.
using master branch at hash c07410d78c2835a82c75650a438eb9d8f0cf2f77

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

lrusinowicz, 2019-02-25:
@orenc17 
Could you please add replication steps (i.e. how do I run these tests)?

orenc17, 2019-02-25:
```
mbed test --compile -m FUTURE_SEQUANA_M0_PSA  -n *psa-* --profile debug
mbed test --compile -m FUTURE_SEQUANA_PSA  -n *psa-* --profile debug -DNO_GREENTEA
```

you can do it on master

0xc0170, 2019-03-14:
@lrusinowicz have you been able to reproduce this?

orenc17, 2019-03-14:
@lrusinowicz please see the latest changes in CY8CKIT_062_WIFI_BT target maybe they can solve the issue on your board as well

lrusinowicz, 2019-03-14:
@orenc17 do you have any particular changes in mind? Do this tests work on CY8CKIT_062_WIFI_BT? Since when?
Unfortunately, I don't have this kit to verify.

MarceloSalazar, 2019-03-14:
@orenc17 please can you point us to the code you're referring to?

orenc17, 2019-03-14:
@lrusinowicz @MarceloSalazar 
cypress added support for PSoC6 wifi and ble kits in PR #9908 
they also implemented a PSA target for the wifi-kit that does not have this issue

i dont know the exact way that they have done that, but its there and it works

lrusinowicz, 2019-03-15:
@orenc17 
Unfortunately, this doesn't move me forward. Cypress is using different PDL version and they have decided to restructure HAL layer significantly, so comparison is difficult. And I cannot verify how their stuff is working (e,g. under debugger) because I don't have the hardware.

orenc17, 2019-03-15:
So I think the solution should be removing mbed_tickless and marking the sleep tests as known issue on future_sequana_psa

@0xc0170 is this an acceptable solution?

cmonr, 2019-03-15:
CC @adbridge @ChiefBureaucraticOfficer

0xc0170, 2019-03-15:
> So I think the solution should be removing mbed_tickless and marking the sleep tests as known issue on future_sequana_psa

tickless: I could not find a reference to this statement (first comment states removing tickless support does not help), - how is tickless related to this issue?

sleep tests: this issue for psa tests? 

Please clarify.

lrusinowicz, 2019-03-15:
I could not reproduce the situation stated in the first comment. For me PSA tests only fail when tickless mode is enabled. Tickless was enabled for FUTURE_SEQUANA_M0_PSA in #9678 to fix issue #9094 - sleep tests failing.

lrusinowicz, 2019-04-10:
It seems that the recent fixes (probably #10291 fixing LP_TICKER on M0) have resolved this issue.
I have run PSA tests using recent master (bb0baee38101738a647c18a7fe8f44a630ad11ab) and almost all tests are passing.
[tests_sequana_psa_gcc_20190410.log](https://github.com/ARMmbed/mbed-os/files/3063253/tests_sequana_psa_gcc_20190410.log)

@orenc17 could you check if you still see the issue?

orenc17, 2019-04-10:
in features\storage\kvstore\conf\global\mbed_lib.json add the following entry:
```
"FUTURE_SEQUANA_PSA": {
            "storage_type": "TDB_INTERNAL"
        }
```

orenc17, 2019-04-10:
#10368  should fix that

MarceloSalazar, 2019-05-09:
Closing - please reopen if continues to be an issue.

closed 2019-05-09 by MarceloSalazar