ARMmbed/mbed-os #9729: mbed compile should output the final flashable image size

teetak01, 2019-02-15:
### Description

Currently Mbed OS compilation output information does not consider other components that increase the binary size, for. ex. SOTP or bootloader, which are part of Device Management Client image.

Current output looks like this:
```
Total Flash memory (text + data): 354257 (+0) bytes
```
Instead it could output something like this:
```
Total Flash memory (text + data): 354257 (+0) bytes
Final Flash image size (Bootloader + other + Application): 395217 (+0) bytes
```

Effectively we could immediately see how much it actually takes space on the device.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

JanneKiiskila, 2019-02-15:
Should it also take into account the flash sector sizes, as those will somewhat limit your life, too?

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@teetak01 Please reopen if you still feel strongly about your proposal.