ARMmbed/mbed-os #9731: No way to set ip stack type requested in AT_CellularContext

Jaco-Liebenberg, 2019-02-15:
### Description

This relates to any target that implements a CellularDevice that inherits from AT_CellularContext but do not override the `connect()` or `do_connect()` functions.

This relates to mbed-os 5.11 and possibly others.

I think there has been a oversight in the implementation of the AT_CellularContex class. In AT_CellularContex there is a private var called `nsapi_ip_stack_t _ip_stack_type_requested;`. `_ip_stack_type_requested` is used by 2 functions in AT_CellularContext. `bool AT_CellularContext::set_new_context(int cid)` and `bool AT_CellularContext::get_context()`. The only place where `_ip_stack_type_requested` gets set, however, is in the constructor of AT_CellularContext. There it gets explicitly set to DEFAULT_STACK. The fact that `_ip_stack_type_requested` is private and that the only place inside AT_CellularContex where `_ip_stack_type_requested` gets set is the constructor, means that `_ip_stack_type_requested` will always be DEFAULT_STACK with no way of changing it to anything else (except of course by changing AT_CellularContext itself). What am I missing?

Clearly AT_CellularContex has been designed with `_ip_stack_type_requested` being any of the stacks listed in `nsapi_ip_stack_t` in mind. Why use a variable if the variable will always have the same value? What I would suggest (and where I think the oversight came in) is to implement a protected or public function that sets `_ip_stack_type_requested` to a desired value or to not use a variable at all but to allow `_ip_stack_type_requested` to be defined through the configuration system and thus ultimately allowing the application designer to define the stack type to use through the use of mbed_app.json.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

lauri-piikivi, 2019-03-21:
Hi,
is there a feature that is affected by this bug/feature?

It is a bit awkward, but there have been modems that only supported IPv4, and in those cases the driver was inherited and stack type overridden. It was not really meant for application API to explicitly set IP version used. Is there a need for such a method?

cmonr, 2019-03-26:
@ARMmbed/mbed-os-ipcore

paspik, 2019-03-26:
What would be the use case for setting IP version used? I would like to understand the need for such a feature.

kjbracey-arm, 2019-03-26:
The issue is more generic than cellular - `NetworkInterface` doesn't expose an IPv4/IPv6 choice itself, which would be why `CellularContext`'s ability to specify isn't used. Same applies for various underlying other APIs which have the setting, like `OnboardNetworkStack`.

If we wanted to expose this, I think it should logically be via `NetworkInterface`.

AriParkkila, 2019-06-06:
@Jaco-Liebenberg Do you mean something like [LWIP::get_ip_addr](https://github.com/ARMmbed/mbed-os/blob/2b354a40b068147f7ccddf15204d1e77a5b88ba5/features/lwipstack/lwip_tools.cpp#L126), which uses `IP_VERSION_PREF` to select IP version? Currently Mbed cellular prefers always to IPv6 in `CellularStack::get_ip_address`.

I think that `_ip_stack_type_requested` was originally to request a specific IP version from a cellular network, but in real that's unfeasible due to cellular networks may disregard whatever clients request.

0xc0170, 2019-08-14:
@Jaco-Liebenberg Any update to the latest comments?

Jaco-Liebenberg, 2019-08-19:
Hi sorry for being so silent this long. I got distracted at work and did not do any mbed-os related work for a while now. 

I opened this issue in relation to the design of the AT_CellularContext class as it existed in mbed-os 5.11.

Let me see if I can elaborate a bit more. I came across this issue while implementing a driver for a modem not yet supported by mbed-os. I started implementing the driver by inheriting from AT_CellularContext. AT_CellularContext has (amongst others) the following two virtual functions: `connect()` and `do_connect()`. Marking a function as a virtual function (as opposed to a pure virtual function) suggests (at least in my experience) that whoever inherits from that class should decide whether or not they are happy with the implementation of that function and override it if they are not. This allows for cleaner, simpler code with less code duplication. Implementing the driver for this new modem, I was happy with the implementation of the `connect()` function in AT_CellularContext and as such I did not override it. The `connect()` function calls the  `do_connect()` function. The practical implication here (as well as the design behind it) being to allow the children of AT_CellularContext to not override `connect()` but use it as is and only override `do_connect`. This allows for the children of AT_CellularContext to leave the connect function as is (and as such not create any duplicate code) but only change a little part of it; the `do_connect` part.

The `do_connect` function again is marked as virtual with its own base implementation in AT_CellularContext. As with the `connect()` function I was happy with the base implementation if the `do_connect` function and did not feel the need to override it in the implementation of my driver. The problem, however, came in with the fact that the base implementation of the `do_connect()` function calls the private function `do_activate_context()` which in turn calls the private function `get_context()`.

Let's analyse the following part of the `get_context()` function: 
```
if (_ip_stack_type_requested == IPV4_STACK) {
    if (pdp_stack == IPV4_STACK || pdp_stack == IPV4V6_STACK) {
        _ip_stack_type = _ip_stack_type_requested;
        _cid = cid;
        break;
    }
} else if (_ip_stack_type_requested == IPV6_STACK) {
    if (pdp_stack == IPV6_STACK || pdp_stack == IPV4V6_STACK) {
        _ip_stack_type = _ip_stack_type_requested;
        _cid = cid;
        break;
      }
} else {
    // requested dual stack or stack is not specified 
    ...
```

Here `_ip_stack_type_requested` gets tested for IPV4_STACK and IPV6_STACK but because `_ip_stack_type_requested` is private and because the only place where `_ip_stack_type_requested` ever gets set is in the constructor, the code above will thus always end up at the "requested dual stack or stack is not specified" part because `_ip_stack_type_requested` will always == `DEFAULT_STACK`. 
 The only place where AT_CellularContext sets the value for `_ip_stack_type_requested` is in its constructor and here it gets set to `DEFAULT_STACK` . No where else in AT_CellularContext does the value for `_ip_stack_type_requested` change and because it is private none of the children of AT_CellularContext can change its value. Here I should maybe state that ` _ip_stack_type_requested` is of type `nsapi_ip_stack_t` and `nsapi_ip_stack_t` has the following definition: 
```
/** Supported IP protocol versions of IP stack
 *
 *  @enum nsapi_ip_stack
 */
typedef enum nsapi_ip_stack {
    DEFAULT_STACK = 0,
    IPV4_STACK,
    IPV6_STACK,
    IPV4V6_STACK
} nsapi_ip_stack_t;
```
The other two parts (the `_ip_stack_type_requested == IPV4_STACK` and ` _ip_stack_type_requested == IPV6_STACK` parts) of the if-else statement is practically unreachable code. It is here where I feel the bug in the design of AT_CellularContext crept in.

Standing back a bit and looking at the overall design of the AT_CellularContext class (looking at the base implementation of the `connect()` and `do_connect` functions in conjunction with the `get_context()` function) it seems to me that the designer of AT_CellularContext intended for ` _ip_stack_type_requested` to change at run time either directly by the children of AT_CellularContext or by code implementing or using the children of AT_CellularContext (maybe that functionality did once exist but were removed in subsequent mbed-os versions?) It is from this that my suggestion for a public or private function to modify the value of ` _ip_stack_type_requested` came from.

I personally feel that `_ip_stack_type_requested ` will at least be known at compile time of the main Application if not already known when the driver for the particular modem is implemented. It is from this that my second suggestion to make it configurable via the configuration system came from.

I do see though that the design of AT_CellularContext between mbed-os version 5.11 and 5.13 has changed and that `_ip_stack_type_requested ` do not exist in 5.13 anymore so I doubt this issue is relevant anymore.

kivaisan, 2019-09-11:
I think this is no longer an issue. Selecting stack type has been modified since 5.11 a lot and there is still one more fix coming up (https://github.com/ARMmbed/mbed-os/pull/11424) which selects the stack type based on the IP address got from the network.
But generally I think we could change visibility of some internal methods as protected to enable easier overriding of `connect`/`do_connect`.

closed 2019-09-11 by ciarmcom