ARMmbed/mbed-os #9732: CordioBLE stack Read callback for RW attributes

akkishore1, 2019-02-15:
### Description

We found that that when a GATT attribute is configured as Read/Write by the Gattserver, the application does not receive Read callback when GATT client tries to read the attribute. While debugging the issue we found following issue in Cordio stack implementation.

In file CordioGattserver.cpp in fxn GattServer::insert_characteristic_value_attribute the call back bit is over written as assignment is done instead of Bitwise or.  After changing to bitwise or then we are getting the read callbacks for Readwrite attributes. Is this a bug or intentional implementation?

// Set value attribute settings
    if (properties & READ_PROPERTY) {
        attribute_it->settings = ATTS_SET_READ_CBACK;
    }
    if (properties & WRITABLE_PROPERTIES) {
        attribute_it->settings = ATTS_SET_WRITE_CBACK;    // Change this to         attribute_it->settings |= ATTS_SET_WRITE_CBACK
    }

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

desmond-blue, 2019-02-19:
#9748

akkishore1, 2019-03-14:
Issue is resolved in 5.12.0

closed 2019-03-14 by akkishore1