ARMmbed/mbed-os #9733: CordioBLE stack MTU size configuration

akkishore1, 2019-02-15:
### Description

The Homekit BLE implementation requires MTU size of 150 or larger. But in Cordio stack the MTU size is hardcoded to 23 and hence MTU negotiation seems to be failing. Is it possible to increase MTU size in Cordio stack

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

j3hill, 2019-02-15:
@akkishore1 Thank for submitting this issue. Increasing the MTU size is a feature request, and I'm updating this Issue to reflect that.

pan-, 2019-02-19:
@akkishore1 We're adding compile time configuration of the Cordio stack for bed OS 5.12. I will update this issue when the PR is up (few days).

donatieng, 2019-03-01:
Hi @akkishore1,

Both of these PRs are now in:
* https://github.com/ARMmbed/mbed-os/pull/9813 will let you configure the desired MTU for Cordio
* https://github.com/ARMmbed/mbed-os/pull/9537 will provide you with notifications for MTU negotiation

akkishore1, 2019-03-14:
MTU configuration support addes to JSON file

closed 2019-03-14 by akkishore1