ARMmbed/mbed-os #9734: Question: Utest/Unity Correct Output on Target Serial

nuertey, 2019-02-15:
### Description

Hello,
I have a unit test application (baselined off of mbed-os-5.11.4) that I have built with mbed-cli, GCC_ARM and are manually running (i.e. without Greentea) on my NUCLEO_F767ZI target. 

Is there a recommended setup procedure (test configuration file params? framework lib configuration file params?) that I must follow in order to ensure that the test output to the serial does not hardfault? 

What about the CR/LF misalignment test output in a few cases that I do observe? Any similar procedures to properly control those? 

I am thinking something alongst the lines of these configuration parameters when building a proper application :
```
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 9600,
            "platform.stdio-convert-newlines": true
        }
    }
```
Thanks
Nuertey

### Issue request type

    [X ] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-02-15:
> ensure that the test output to the serial does not hardfault?

Why would it hardfault?

General rules guidelines: `platform.stdio-convert-newlines` should be true (and the app should just be using `'\n'` in its output). `platform.stdio-buffered-serial` should be true, baud rate can be whatever you like.

Don't make your own `Serial` or `UARTSerial` object on USBTX/USBRX. Just use `stdin` and `stdout`. 

Output should not be performed from IRQ handlers.

nuertey, 2019-02-15:
Thanks @kjbracey-arm for the guidelines. I was observing the dreaded blinking LED so had assumed that it was due to the serial misbehaving. Perhaps that was rather due to my last-run testcase failing? In other words if a testcase's assert fails and the test application subsequently ends in failure, will that LED blink?

Also, note that after following your guidelines and defining those parameters in my mbed_app.json, I still observe some but not all of Utest/Unity's output misaligned. Could that be some subtle bug and the particular lines might provide a clue in resolving it? Here is my output, and as you can see, it makes reading tedious. Thanks. :
```
>>> Running 30 test cases...
                            {{__testcase_name;Testing base 64 decoding}}
{{__testcase_name;Testing base 64 URL decoding}}
{{__testcase_name;Testing base 64 encoding}}
{{__testcase_name;Testing base 64 URL encoding}}
{{__testcase_name;Testing token format - Missing Dot}}
{{__testcase_name;Testing token format - Invalid Character}}
{{__testcase_name;Testing token format - Invalid JSON}}
{{__testcase_name;Testing decoding of a token}}
{{__testcase_name;Testing creation of a token}}
{{__testcase_name;Testing creation of an HS256 token}}
{{__testcase_name;Testing creation of an RS256 token}}
{{__testcase_name;Testing creation of an RS512 token}}
{{__testcase_name;Testing creation of a PS256 token}}
{{__testcase_name;Testing creation of a PS384 token}}
{{__testcase_name;Testing creation of a PS512 token}}
{{__testcase_name;Testing creation of an ES256 token}}
{{__testcase_name;Testing creation failure of an ES256 token with no private key}}
{{__testcase_name;Testing verification of an RS256 token}}
{{__testcase_name;Testing verification of an RS256 token with only a public key}}
{{__testcase_name;Testing verification failure of an RS256 token with invalid public key}}
{{__testcase_name;Testing verification of an RS512 token}}
{{__testcase_name;Testing verification of an RS512 token with only a public key}}
{{__testcase_name;Testing verification failure of an RS512 token with invalid public key}}
{{__testcase_name;Testing verification of an HS256 token}}
{{__testcase_name;Testing verification failure in the presence of incorrect claim fields}}
{{__testcase_name;Testing verification of an ES256 token}}
{{__testcase_name;Testing verification failure of an ES256 token with invalid public key}}
{{__testcase_name;Testing verification of a PS256 token}}
{{__testcase_name;Testing verification of a PS256 token with only a public key}}
{{__testcase_name;Testing verification failure of a PS256 token with invalid public key}}

>>> Running case #1: 'Testing base 64 decoding'...
                                                  {{__testcase_start;Testing base 64 decoding}}
[expected outcome] system :-> "Success"
[actual errorCode] system :-> "Success"
[decoded.c_str] :-> "1"
ting base 64 decoding': 1 passed, 0 failed

                                          >>> Running case #2: 'Testing base 64 URL decoding'...
                                                                                                {{__testcase_start;Testing base 64 URL decoding}}
{{__testcase_finish;Testing base 64 URL decoding;1;0}}
>>> 'Testing base 64 URL decoding': 1 passed, 0 failed

                                                      >>> Running case #3: 'Testing base 64 encoding'...
                                                                                                        {{__testcase_start;Testing base 64 encoding}}
{{__testcase_finish;Testing base 64 encoding;1;0}}
>>> 'Testing base 64 encoding': 1 passed, 0 failed

                                                  >>> Running case #4: 'Testing base 64 URL encoding'...
                                                                                                        {{__testcase_start;Testing base 64 URL encoding}}
{{__testcase_finish;Testing base 64 URL encoding;1;0}}
>>> 'Testing base 64 URL encoding': 1 passed, 0 failed

                                                      >>> Running case #5: 'Testing token format - Missing Dot'...
                                                                                                                  {{__testcase_start;Testing token format - Missing Dot}}
JWT-Mbed :-> "Invalid token argument supplied"
JWT-Mbed :-> "Invalid token argument supplied"
g Dot;1;0}}
>>> 'Testing token format - Missing Dot': 1 passed, 0 failed

                                                            >>> Running case #6: 'Testing token format - Invalid Character'...
                                                                                                                              {{__testcase_start;Testing token format - Invalid Character}}
JWT-Mbed :-> "Runtime error. Invalid input"
esting token format - Invalid Character;1;0}}
>>> 'Testing token format - Invalid Character': 1 passed, 0 failed
[... 1 lines cut ...]
                                                                  >>> Running case #7: 'Testing token format - Invalid JSON'...
                                                                                                                               {{__testcase_start;Testing token format - Invalid JSON}}
JWT-Mbed :-> "Invalid JSON detected error"
Testing token format - Invalid JSON;1;0}}
>>> 'Testing token format - Invalid JSON': 1 passed, 0 failed

[... cut ...]
                                                                                    >>> Test cases: 8 passed, 1 failed with reason 'Assertion Failed'
                                                                                                                                                     >>> TESTS FAILED!
                                                                                                                                                                      {{__testcase_summary;8;1}}
{{max_heap_usage;7165}}
{{reserved_heap;477200}}
{{end;failure}}
{{__exit;0}}
```

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@nuertey You'll observe LED blinking if test fails. Please reopen this issue if you have further questions.