ARMmbed/mbed-os #9735: ATHandler set/restore at_timeout needs lock

NeilMacMullen, 2019-02-15:
### Description
ATHandler::set/restore _at_timeout can collide with the _at_timeout manipulation in process_oob leading to the _at_timeout value being reset (by process_oob) to a prior value.

The fix is to place lock/unlock on entry/exit of these two functions.

While I'm talking about timeouts, the current design with huge #defined values (TIMEOUT_CONNECT is 60 seconds!) is completely inappropriate for a target which wants to power up the modem, squirt a bit of data then power down again.  (We aim to do this in less than 30 seconds - if we fail that's fine - we can try again tomorrow.)  Please consider making these timeouts dynamically tunable.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

AriParkkila, 2019-03-13:
@NeilMacMullen the lock/unlock bug fix is now merged. PR to make timeouts configurable is pending for approval/merge.

By default `CellularContext::connect` retries to connect for ten times, that is far over ten minutes, before giving up and that might be too long in your case... Do you think retry count should be configurable too (via `CellularDevice API`)?

NeilMacMullen, 2019-03-13:
@AriParkkila   Thanks! I worked around the timeout issue by adding a 'backdoor' to AT_CellularContext...
There's probably a better way to do it (and I'm aware the modification as it stands is misleading as it overrides _all_ state timeouts).

Retry count probably should be configurable but I don't think it's the best way to solve the particular problem here.  The use-case can best be summarised as....

* Power up the radio, 
* Try to connect (register with network) for a max of 'n' seconds (n >30) 
* If successful, set up socket, send a couple of K of data, use socket-timeout to give up if data-transmission isn't working well.
* If not successful, just try again tomorrow when device has probably moved to a better coverage area.
* Power off the radio

So really an API that just gives a simple "give up after n seconds regardless of current state" is what is wanted.

```
uint32_t AT_CellularContext::get_timeout_for_operation(ContextOperation op) const
{
      //first check to see if user has overridden current timeout...
	if (_connectionTimeout != 0)
		return  _connectionTimeout;

    uint32_t timeout = NETWORK_TIMEOUT; // default timeout is 30 minutes as registration and attach may take time
    if (op == OP_SIM_READY || op == OP_DEVICE_READY) {
        timeout = DEVICE_TIMEOUT; // use 5 minutes for device ready and sim
    }
    return timeout;
}

 //in cellularcontext.h
void set_connection_timeout(int seconds)
{
	_connectionTimeout = seconds*1000;
}

```

AriParkkila, 2019-03-13:
@NeilMacMullen you probably could connect asynchronously by setting `NetworkInterface::set_blocking(false);` and then wait for connection timeout in the application?

NeilMacMullen, 2019-03-13:
@AriParkkila  - yes I did try that but couldn't get it to work for reasons that I now can't recall - something to do with not being able to correctly monitor the interim state to find out when the connection had completed.   Possibly due to my misunderstanding of how it was intended to work rather than a fault in the stack.  If you can post a few lines showing the canonical sequence of API calls for this I can give it a try again.  ISTR I tried something like...
```
iface->set_blocking(false);
iface->connect();
while (!iface->is_connected() && ! timer.read_ms() < CONNECTION__TIMEOUT) 
{ wait_ms(1000);}
```

AriParkkila, 2019-03-13:
@NeilMacMullen I guess it should work if you wait `while (iface->get_connection_status() != NSAPI_STATUS_GLOBAL_UP)`.

However, network callback would be the best way to save power. Pseudo-code would be something like below.
```
rtos::Semaphore connectSemaphore;
static void network_callback(nsapi_event_t ev, intptr_t ptr)
{
    if (ev == NSAPI_EVENT_CONNECTION_STATUS_CHANGE && ptr == NSAPI_STATUS_GLOBAL_UP) {
            tr_info("Network callback: Connected to a network");
            connectSemaphore.release();
    }
}

bool do_connect()
{
    if (iface->get_connection_status() != NSAPI_STATUS_GLOBAL_UP) {
        iface->attach(callback(&network_callback));
        iface->set_blocking(false);
        if (iface->connect() == NSAPI_ERROR_OK) {
            if (connectSemaphore.wait(30*1000) > 0) { // wait connect for 30 seconds
                return true;
```

NeilMacMullen, 2019-03-14:
Thanks @AriParkkila  - will give this a try in the next week or so and let you know of any problems.

AriParkkila, 2019-04-04:
@NeilMacMullen can we close this issue?

There are now fixes:
- #9751 Cellular: Fix ATHandler set/restore_at_timeout with mutex lock
- #10053 Cellular: Make CellularStateMachine timeouts configurable
- See comment about how to cancel connect after some timeout

closed 2019-05-10 by ciarmcom